acmerobotics/road-runner #13: GUI feature requests.

David10238, 2019-05-24:
1) support for mecanum paths and heading interporlaters.
2) the ability to define the width and length of the bot to see an outline of the bot's path similar to PathWeaver GUI, to help work around obstacles.

rbrott, 2019-05-25:
Agreed. A new GUI and trajectory file format will be in the works soon. At the moment, I'm leaning toward switching over to FX and dropping the plugin (which requires Swing).

rbrott, 2019-12-18:
I decided to give the old Swing GUI some love instead of building a separate FX one (I still like the idea of the plugin). Mecanum constraints (including the new lateral multiplier) and heading interpolators are supported. Your second item is on the list.

closed 2020-03-14 by rbrott