acmerobotics/road-runner #15: Plugin occasionally with negative numbers

joannareese, 2019-09-09:
This is a stacktrace output: 
`java.lang.ArrayIndexOutOfBoundsException`
	at java.lang.System.arraycopy(Native Method)
	at kotlin.collections.ArraysKt___ArraysJvmKt.copyInto(_ArraysJvm.kt:796)
	at kotlin.collections.ArraysKt___ArraysJvmKt.copyInto$default(_ArraysJvm.kt:795)
	at com.acmerobotics.roadrunner.path.Path.reparam$core(Path.kt:138)
	at com.acmerobotics.roadrunner.trajectory.TrajectoryGenerator$generateProfile$1.get(TrajectoryGenerator.kt:34)
	at com.acmerobotics.roadrunner.profile.MotionProfileGenerator.generateMotionProfile(MotionProfileGenerator.kt:335)
	at com.acmerobotics.roadrunner.trajectory.TrajectoryGenerator.generateProfile(TrajectoryGenerator.kt:23)
	at com.acmerobotics.roadrunner.trajectory.TrajectoryGenerator.generateTrajectory(TrajectoryGenerator.kt:103)
	at com.acmerobotics.roadrunner.trajectory.TrajectoryBuilder.buildTrajectory(TrajectoryBuilder.kt:49)
	at com.acmerobotics.roadrunner.trajectory.BaseTrajectoryBuilder.build(BaseTrajectoryBuilder.kt:163)
	at com.acmerobotics.roadrunner.trajectory.TrajectoryConfig.toTrajectory(TrajectoryConfig.kt:54)
	at com.acmerobotics.roadrunner.gui.MainPanel.updateTrajectory(MainPanel.kt:66)
	at com.acmerobotics.roadrunner.gui.MainPanel.access$updateTrajectory(MainPanel.kt:23)
	at com.acmerobotics.roadrunner.gui.MainPanel$1.invoke(MainPanel.kt:38)
	at com.acmerobotics.roadrunner.gui.MainPanel$1.invoke(MainPanel.kt:23)
	at com.acmerobotics.roadrunner.gui.PoseEditorPanel.fireUpdate(PoseEditorPanel.kt:59)
	at com.acmerobotics.roadrunner.gui.PoseEditorPanel.access$fireUpdate(PoseEditorPanel.kt:12)
	at com.acmerobotics.roadrunner.gui.PoseEditorPanel$addPose$1.invoke(PoseEditorPanel.kt:80)
	at com.acmerobotics.roadrunner.gui.PoseEditorPanel$addPose$1.invoke(PoseEditorPanel.kt:12)
	at com.acmerobotics.roadrunner.gui.SwingKt$addChangeListener$1.removeUpdate(swing.kt:11)
	at javax.swing.text.AbstractDocument.fireRemoveUpdate(AbstractDocument.java:259)
	at javax.swing.text.AbstractDocument.handleRemove(AbstractDocument.java:622)
	at javax.swing.text.AbstractDocument.remove(AbstractDocument.java:590)
	at com.intellij.openapi.editor.textarea.TextComponentDocument.deleteString(TextComponentDocument.java:99)
	at com.intellij.openapi.editor.actions.BackspaceAction.doBackSpaceAtCaret(BackspaceAction.java:67)
	at com.intellij.openapi.editor.actions.BackspaceAction.access$100(BackspaceAction.java:17)
	at com.intellij.openapi.editor.actions.BackspaceAction$Handler.executeWriteAction(BackspaceAction.java:36)
	at com.intellij.openapi.editor.actionSystem.EditorWriteActionHandler$1.run(EditorWriteActionHandler.java:51)
	at com.intellij.openapi.editor.actionSystem.EditorWriteActionHandler.doExecute(EditorWriteActionHandler.java:62)
	at com.intellij.openapi.editor.actionSystem.EditorActionHandler.execute(EditorActionHandler.java:219)
	at com.intellij.codeInsight.editorActions.BackspaceHandler.executeWriteAction(BackspaceHandler.java:48)
	at com.intellij.openapi.editor.actionSystem.EditorWriteActionHandler$1.run(EditorWriteActionHandler.java:51)
	at com.intellij.openapi.editor.actionSystem.EditorWriteActionHandler.doExecute(EditorWriteActionHandler.java:62)
	at com.intellij.openapi.editor.actionSystem.EditorActionHandler.lambda$null$2(EditorActionHandler.java:209)
	at com.intellij.openapi.editor.actionSystem.EditorActionHandler.doIfEnabled(EditorActionHandler.java:103)
	at com.intellij.openapi.editor.actionSystem.EditorActionHandler.lambda$execute$3(EditorActionHandler.java:208)
	at com.intellij.openapi.editor.textarea.TextComponentCaretModel.runForEachCaret(TextComponentCaretModel.java:235)
	at com.intellij.openapi.editor.actionSystem.EditorActionHandler.execute(EditorActionHandler.java:206)
	at com.intellij.openapi.editor.actions.DeleteSelectionHandler.executeWriteAction(DeleteSelectionHandler.java:46)
	at com.intellij.openapi.editor.actionSystem.EditorWriteActionHandler$1.run(EditorWriteActionHandler.java:51)
	at com.intellij.openapi.editor.actionSystem.EditorWriteActionHandler.doExecute(EditorWriteActionHandler.java:62)
	at com.intellij.openapi.editor.actionSystem.EditorActionHandler.lambda$execute$4(EditorActionHandler.java:216)
	at com.intellij.openapi.editor.actionSystem.EditorActionHandler.doIfEnabled(EditorActionHandler.java:103)
	at com.intellij.openapi.editor.actionSystem.EditorActionHandler.execute(EditorActionHandler.java:215)
	at com.intellij.codeInsight.lookup.impl.BackspaceHandler.doExecute(BackspaceHandler.java:25)
	at com.intellij.openapi.editor.actionSystem.EditorActionHandler.lambda$execute$4(EditorActionHandler.java:216)
	at com.intellij.openapi.editor.actionSystem.EditorActionHandler.doIfEnabled(EditorActionHandler.java:103)
	at com.intellij.openapi.editor.actionSystem.EditorActionHandler.execute(EditorActionHandler.java:215)
	at com.intellij.openapi.editor.actionSystem.EditorAction.lambda$actionPerformed$0(EditorAction.java:102)
	at com.intellij.openapi.command.impl.CoreCommandProcessor.executeCommand(CoreCommandProcessor.java:220)
	at com.intellij.openapi.command.impl.CoreCommandProcessor.executeCommand(CoreCommandProcessor.java:178)
	at com.intellij.openapi.editor.actionSystem.EditorAction.actionPerformed(EditorAction.java:111)
	at com.intellij.openapi.editor.actionSystem.EditorAction.actionPerformed(EditorAction.java:86)
	at com.intellij.openapi.actionSystem.ex.ActionUtil$1.run(ActionUtil.java:261)
	at com.intellij.openapi.actionSystem.ex.ActionUtil.performActionDumbAware(ActionUtil.java:278)
	at com.intellij.openapi.keymap.impl.IdeKeyEventDispatcher$1.performAction(IdeKeyEventDispatcher.java:625)
	at com.intellij.openapi.keymap.impl.IdeKeyEventDispatcher.lambda$processAction$2(IdeKeyEventDispatcher.java:674)
	at com.intellij.openapi.application.TransactionGuardImpl.performUserActivity(TransactionGuardImpl.java:195)
	at com.intellij.openapi.keymap.impl.IdeKeyEventDispatcher.processAction(IdeKeyEventDispatcher.java:673)
	at com.intellij.openapi.keymap.impl.IdeKeyEventDispatcher.processActionOrWaitSecondStroke(IdeKeyEventDispatcher.java:518)
	at com.intellij.openapi.keymap.impl.IdeKeyEventDispatcher.inInitState(IdeKeyEventDispatcher.java:473)
	at com.intellij.openapi.keymap.impl.IdeKeyEventDispatcher.dispatchKeyEvent(IdeKeyEventDispatcher.java:223)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:681)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:375)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)'
Will also crash android studio as a whole

rbrott, 2019-09-10:
Looks like there is an issue with generating the motion profile for the input path. Could you provide the poses and constraints that you provided? (or attach the trajectory YAML file)

joannareese, 2019-09-10:
The issue happened randomly while making these files:
 https://drive.google.com/drive/folders/1e1TeUhpKHzM4P6SUtQGKeHlHQmVrj37f?usp=sharing

rbrott, 2019-09-10:
OK I was able to reproduce by putting two successive poses close together (finer than the resolution).

joannareese, 2019-09-10:
Ok good to know for now I think we will just increase the resolution

rbrott, 2019-09-14:
Turns out this issue was a symptom of a larger problem with the reparametrization of reversed paths. It should be fixed in https://github.com/acmerobotics/road-runner/commit/e4451432ae6aad5e5299d6448a7f84033f7922f2.

closed 2019-09-14 by rbrott