acmerobotics/road-runner #19: Exception "java.lang.IndexOutOfBoundsException: sep outside of the progression"

LetianLi, 2019-12-19:
When calling buildTrajectory, I often got exception "java.lang.IndexOutOfBoundsException: sep outside of the progression". I sometimes got this error, sometimes not, with exact same code and same parameters.

What does this mean? And how to prevent it?

Exception with following call stack:

2019-12-18 21:58:53.856 8535-8943/com.qualcomm.ftcrobotcontroller E/EventLoopManager: java.lang.IndexOutOfBoundsException: sep outside of the progression
2019-12-18 21:58:53.858 8535-8943/com.qualcomm.ftcrobotcontroller E/EventLoopManager:     at com.acmerobotics.roadrunner.util.DoubleProgression.split(DoubleProgression.kt:69)
2019-12-18 21:58:53.860 8535-8943/com.qualcomm.ftcrobotcontroller E/EventLoopManager:     at com.acmerobotics.roadrunner.path.Path.reparam$core(Path.kt:135)
2019-12-18 21:58:53.861 8535-8943/com.qualcomm.ftcrobotcontroller E/EventLoopManager:     at com.acmerobotics.roadrunner.trajectory.TrajectoryGenerator$generateProfile$1.get(TrajectoryGenerator.kt:34)
2019-12-18 21:58:53.863 8535-8943/com.qualcomm.ftcrobotcontroller E/EventLoopManager:     at com.acmerobotics.roadrunner.profile.MotionProfileGenerator.generateMotionProfile(MotionProfileGenerator.kt:335)
2019-12-18 21:58:53.865 8535-8943/com.qualcomm.ftcrobotcontroller E/EventLoopManager:     at com.acmerobotics.roadrunner.trajectory.TrajectoryGenerator.generateProfile(TrajectoryGenerator.kt:23)
2019-12-18 21:58:53.866 8535-8943/com.qualcomm.ftcrobotcontroller E/EventLoopManager:     at com.acmerobotics.roadrunner.trajectory.TrajectoryGenerator.generateTrajectory(TrajectoryGenerator.kt:103)
2019-12-18 21:58:53.868 8535-8943/com.qualcomm.ftcrobotcontroller E/EventLoopManager:     at com.acmerobotics.roadrunner.trajectory.TrajectoryBuilder.buildTrajectory(TrajectoryBuilder.kt:49)
2019-12-18 21:58:53.869 8535-8943/com.qualcomm.ftcrobotcontroller E/EventLoopManager:     at com.acmerobotics.roadrunner.trajectory.BaseTrajectoryBuilder.build(BaseTrajectoryBuilder.kt:163)

rbrott, 2019-12-19:
This is a known issue with super short path segments. It's good to know someone has encountered it in the wild. One workaround is to ensure that no two adjacent path/trajectory builder positions coincide (thus attempting to create a zero-length path segment).

LetianLi, 2019-12-19:
How short is considered to be too short?

rbrott, 2019-12-19:
Too short is within a few multiples of the resolution which defaults to 0.25 (which is sensible for inches). If you're using a different distance unit, you should definitely change the resolution.

commit 50b5c3b8 (2020-01-25): Re-implement DoubleProgression to properly handle small trajectories (fixes #19)
closed 2019-12-22 by LetianLi