acmerobotics/road-runner #22: Roadrunner y-axis Negated

cxqtang, 2020-01-05:
Our robot moves in the wrong y-direction under a translational PID when following a spline, and speeds up at the end due to the increasing error. However, the robot follows the path correctly without a translational PID, meaning the motors are each getting the correct velocity. We're currently using the Roadrunner library, and we're using the SampleREVMecanumOptimized drive base that is slightly modified to account for our drive motors being on two seperate hubs. We're currently using just our drive encoders and the imu for localization, which should be the default setting of the mecanum drive base. I've ran the localization test and it seems that the y-value is increasing when it should be decreasing and vice versa. The x-value is correct. How should I troubleshoot the localization? Is there some way the negate the y-value reading when the robot is following a trajectory?

rbrott, 2020-01-07:
This is very peculiar. Are you using the built-in velocity PID? If you give the motors positive power outside of Road Runner, do the encoder positions/readings increase?

cxqtang, 2020-01-07:
Figured out the bug. For our NeveRest20Orbital motor properties class, we've accidentally set the rotation to clockwise instead of counter-clockwise.

closed 2020-01-07 by cxqtang