acmerobotics/road-runner #23: Preferred method for running "timed" code in TrajectoryBuilder

nchowder, 2020-01-09:
This is a question, not an issue.

Let's say I want to move a servo to position 0 sometime during the trajectory. Easy, I just do `.addMarker(() -> {robot.servo.setPosition(0); return null;}` (using Java). But what if I want to start a routine that moves the servo to position 1, waits 1000ms, then moves to position 0, while continuing the motion trajectory? Something like `robot.servo.setPosition(0); sleep(1000); robot.servo.setPosition(1);` freezes the main thread as well.

My strategy was to create a Runnable object that contains all the operations that should be run according to these timings, then call its `.run()` in `addMarker()`. Is this the right way to go about this problem? Any problems I should watch out for?

rbrott, 2020-01-09:
I didn't anticipate the apparent popularity of markers, so the the marker interface in 0.4.x is incomplete and has some implementation issues. 0.5.0 will hopefully ship with more ways to create markers including what you describe.

For now, I recommend writing some asynchronous state machine logic independent of Road Runner and use markers to trigger state transitions (e.g., to start a timer). You'll lose the ability to use the synchronous methods in the quickstart, but it's more simple and efficient than using a thread.

nchowder, 2020-01-09:
How would I write state logic outside of road runner without a new thread? The trajectory builder is blocking, so I can't really run anything outside of it in the main program. Unless I'm not understanding you well.

rbrott, 2020-01-09:
Trajectory followers are not blocking, but I can understand how it seems that way. In the quickstart `followTrajectorySync()` feeds the trajectory to the follower and repeatedly updates the follower to execute the trajectory. To incorporate other concurrent tasks, you will have to make your own version of this following loop.
```java
timer = ...;
state = ...;
drive.followTrajectory(...);
while (opModeIsActive()) {
    drive.update();

    switch (state) {
        case WAITING_FOR_TIMER:
            if (timer.elapsedTime() > DURATION) {
                // do something
                state = State.IDLE;
            }
            break;
        case IDLE:
            break;
        // ...
    }
}
```
This state machine logic can be completely independent of the trajectory, or you can have some shared mutable state that trajectory markers can use to communicate with the other logic.

nchowder, 2020-01-09:
Ok, I think i get it. Thanks for the help. I'll reply to this issue if I run into any roadblocks.

EddieDL, 2020-01-10:
@nchowder I have done something very similar. If you are using the QuickStart or similar implementation, you could do the following.

Update `waitForIdle` and `followTrajectorySync` to have an additional optional parameter `task`. Which is an implementation of an interface (`ExecuteWhileMoving`) with an `execute(Pose2d)` method. Then I just call `task.execute()` on each loop in `waitForIdle`. NOTE: I am not a Java expert, I would prefer to use some sort of dynamic expressions instead of forcing the implementation of an interface. I just wasn't exactly sure how to do that in Java.

Also, in my `execute()` I don't use time, I use thresholds for where the position of the robot should be. For example:

```
class ActionAtDistance implements ExecuteWhileMoving {
        double max;
        
        public ActionAtDistance(double max) {
            this.max = max;
        }

        public boolean execute(Pose2d pose) {
            if(pose.getX() < max) {
                // Do something
            }
            else if(pose.getX() < max + 40) {
                // Do something else
            }
            else {
              // Do nothing or something :)
           }
            return true;
        }
    }
```
In my version you can actually use the return to tell the system to stop, if you want.

Not sure it helps, but thought I would share.

closed 2020-03-14 by rbrott