acmerobotics/road-runner #26: Best way to get robot's (predicted) position in TrajectoryBuilder?

AdrianVovk, 2020-01-20:
I'm trying to create a simple convenience extension function for TrajectoryBuilder that lets me cut a lot of boilerplate whenever I call `splineTo`. The problem is, I need to know what the robot's position would be once I get to that step in the trajectory. 

So if I start at (0, 0, 0), drive +x 15 inches, drive -y 45 inches, turn +PI radians, and then execute my extension function, how would I get a "poseEstimate" that is (15, -45, PI)? Right now, I'm doing `val poseEstimate = this.build().end()` and this works sometimes but other times it returns (0, 0, 0) even though the robot is positioned at (-22.5, -62, PI/2)

Thank you!

AdrianVovk, 2020-01-22:
I've solved this by building my own local 5.0 snapshot and using it. Closing

closed 2020-01-22 by AdrianVovk

rbrott, 2020-01-22:
I'm not quite sure why 0.5.0 would fix the `TrajectoryBuilder#end()` issue.

AdrianVovk, 2020-01-22:
@rbrott It doesn't, but TrajectoryBuilder.currentPosition does what I need

rbrott, 2020-02-02:
I wrote the following test case, and I couldn't get it to fail.
```java

@Test
fun testTrajectoryEnd() {
    val endPose = Pose2d(25.0, 25.0, 0.0)
    repeat (50) {
        val traj = TrajectoryBuilder(Pose2d(), constraints = DriveConstraints(5.0, 10.0, 0.0, 2.0, 3.0, 0.0))
            .splineTo(Pose2d(50 * Math.random(), 50 * Math.random(), 2 * PI * Math.random()))
            .splineTo(Pose2d(50 * Math.random(), 50 * Math.random(), 2 * PI * Math.random()))
            .splineTo(endPose)
            .build()
        assert(traj.end() epsilonEquals endPose)
    }
}
```
As far as I can tell, `Trajectory#end()` is safe and, in fact, recommended for constructing a series of sequential trajectories.

AdrianVovk, 2020-02-03:
@rbrott It works wonderfully when you are composing things together the way you do in the test case. What doesn't work is this (rough pseudocode):

```
dt.poseEstimate = Pose2d(x = 10, y = 10, heading = 10)
val traj = dt.trajectoryBuider().build()
assert(traj.end() epsilonEquals dt.poseEstimate)
```

rbrott, 2020-02-03:
I see now. `TrajectoryBuilder()` will start throwing exceptions for empty paths and path segments in 0.5.0.