acmerobotics/road-runner #28: Attempt to Invoke Virtual Method com.acmerobotics.dashboard.FtcDashboard.setTelemetryTransmissionInterval(int) on a null object

BeckettOBrien, 2020-01-22:
Hello, I just downloaded the quickstart to test on my team's bot and I always get this error on any of the opcodes:
`Error: User code threw an uncaught exception: NullPointerException - Attempt to invoke virtual method 'void.acmerobotics.dashboard.FtcDashboard.setTelemetryTransmissionInterval(int)' on a null object reference`
I tried to copy the drive and util directories to my current project (and this is probably what caused the issue) so I was wondering if there was any way to copy the quickstart into an existing project. I added the implementations to the build.release.gradle

rbrott, 2020-01-23:
Sounds like FTC dashboard hasn't been started properly. There are additional lines in the main activity required for initialization. See [these instructions](https://acmerobotics.github.io/ftc-dashboard/gettingstarted).

AndrewLuGit, 2020-01-23:
Hi, I have also used FTC Dashboard and gotten that error, even with the additional lines in the main activity. If you have added the lines and are still getting this error, just close the app and reopen it, and the dashboard should start properly. You can check if the dashboard is started or not since a blue line will appear on the main screen of the app showing if the dashboard is enabled or not.

rbrott, 2020-01-23:
Do you know what circumstances cause this issue? Is it still present in the latest release?

AndrewLuGit, 2020-01-23:
This was on version 0.3.4. It seems the issue happens when the robot controller app is crashed and automatically restarts. For example, the robot controller gets a stuck in stop() error and restarts the app; after the app has restarted the blue line disappears, and if you try to start an opmode that uses the dashboard you will get the null pointer exception.

rbrott, 2020-01-23:
Oh ok. But you haven't been able to reproduce with 0.3.6?

AndrewLuGit, 2020-01-23:
Just updated my dashboard version to 0.3.6, and it looks like it is no longer an issue.

closed 2020-01-23 by BeckettOBrien