acmerobotics/road-runner #30: Strafing specific distance..

dvpatel, 2020-01-24:
Hi,

Learning road-runner...  was able to program the robot to drive forward / reverse specific distance, getting good results.  For strafing, not the case.  I'm using the following snippet but the robot falls well below its target.  Could it be I need separate PIDF values for strafing?

        Trajectory trajectory = drive.trajectoryBuilder()
                .strafeRight(48)
                .build();
        drive.followTrajectorySync(trajectory);

rbrott, 2020-01-24:
You probably need to modify the kinematics to provide extra voltage when strafing. This feature is coming in 0.5.0, but the release isn't quite ready yet. In the meantime, you can either backport the new kinematics (from [this commit](https://github.com/acmerobotics/road-runner/commit/8c8036930fd10e18b7c9ff5b59ae38e0ee8e5350)) or modify the localizer to report a shorter distance traveled than specified and have a different set of translational coefficients for the strafe direction.

dvpatel, 2020-01-24:
Thanks for the guidance.  Backported the code, will let you know how it works out.  

As a side note, not necessary an issue, know of any good docs on PIDF tuning?  I followed instructions from this doc:  https://docs.google.com/document/d/1tyWrXDfMidwYyP_5H4mZyVgaEswhOC35gvdmP-V-5hA/mobilebasic

Wasn't clear if each motor needs to have its own PIDF values.

rbrott, 2020-01-24:
>Thanks for the guidance. Backported the code, will let you know how it works out.

:+1:

>As a side note, not necessary an issue, know of any good docs on PIDF tuning? I followed instructions from this doc: https://docs.google.com/document/d/1tyWrXDfMidwYyP_5H4mZyVgaEswhOC35gvdmP-V-5hA/mobilebasic

The tuning heuristics in that file are not sufficient for good performance. Here are some instructions that will hopefully make their way into the proper docs soon:
1. Increase kP until any phase lag is eliminated. Concurrently increase kD as necessary to mitigate oscillations.
2. Add kI (or adjust kF) until the steady state/constant velocity plateaus are reached.
3. Back off kP and kD a little until the response is less oscillatory (but without lag).

>Wasn't clear if each motor needs to have its own PIDF values.

No, one set of coefficients should work for all motors. A motor behaving significantly different from the others usually indicates a hardware/mechanical issue.

dvpatel, 2020-01-25:
Hi Ryan,

Tried the backport but didn't quite work.  Configured the robot to strafe 24 inches but traveled around 18 inches.

-Dipesh

rbrott, 2020-01-25:
Yes, you should tune the `lateralMultiplier`. Based on the information you provided, it should be set to 24 / 18 = 1.33.

dvpatel, 2020-01-25:
No dice...  Feels like I'm still missing a step.

rbrott, 2020-01-25:
OK I re-read your message, and I think I know the problem. `MecanumDrive` is not equipped to take a lateral multiplier in 0.4.x. The additional parameter you're passing to the `MecanumDrive` constructor is for the wheel base. You need to make a new `Drive` subclass that uses the modified kinematics (in other words, the entirety of the commit).

dvpatel, 2020-01-26:
Ok.  I took the latest master..  I need to modify AssetsTrajectoryLoader load method in quick-start to use TrajectoryGroupConfig.  What is the wheelBase constructor parameter for this object?  Is this the wheel radius (or diameter?)

Also drive.trajectoryBuilder().reverse() to be missing.  

-Dipesh

rbrott, 2020-01-26:
>I took the latest master.. 

It's probably easier to just backport the lateral multiplier changes. This release changes a few important user-facing interfaces and isn't quite ready.

> I need to modify AssetsTrajectoryLoader load method in quick-start to use TrajectoryGroupConfig. What is the wheelBase constructor parameter for this object? Is this the wheel radius (or diameter?)

No, it's unrelated to the wheel itself. It describes the length between wheels on the same side of the robot (i.e., with different axles).
[image]

>Also drive.trajectoryBuilder().reverse() to be missing.

Yes, `TrajectoryBuilder#reverse()` has been removed. This is one such change made to the builders.

dvpatel, 2020-01-28:
Hi Ryan,

I completely backported to the commit that has the lateral multiplier change.  Strafing is still not working, tested with with various lateral multiplier values.

Also, turning is not properly working.  Falls well below target value.  I tested with target of 90 degrees but actual was somewhere around 30 degrees.

Lastly, I had to create a new motor type for NeveRest 20 Orbital.  The interface is structured as follows.  These values are directly from the motor specs provided by AndyMark.  Hope this helps.

@MotorType(ticksPerRev = 537.6, gearing = 19.2, maxRPM = 340, orientation = Rotation.CW)
@DeviceProperties(xmlTag = "NeveRest20Orbital", name = "NeveRest 20 Orbital Gearmotor", builtIn = true)
@DistributorInfo(distributor = "AndyMark", model = "am-3637", url = "https://www.andymark.com/products/neverest-orbital-20-gearmotor")
@ExpansionHubPIDFVelocityParams(P=10.0, I=3.0, F=0)
@ExpansionHubPIDFPositionParams(P=10.0)
public interface NeveRest20Orbital {
}

-Dipesh

rbrott, 2020-01-28:
Is your code publicly available?

dvpatel, 2020-01-28:
Sorry, it's not.  I can share specific snippets if it helps.

rbrott, 2020-01-28:
It would be helpful to see your drive class and the new kinematics class.

dvpatel, 2020-01-28:
I modified SampleMecanumDriveBase in road-runner-quickstart.  Here's what constructor looks like.  You advised correctly but I forgot.  Any other class I need to update / modify?

    public SampleMecanumDriveBase() {
        super(kV, kA, kStatic, TRACK_WIDTH, WHEEL_BASE, LATERAL_MULTIPLIER);

        dashboard = FtcDashboard.getInstance();
        dashboard.setTelemetryTransmissionInterval(25);

        clock = NanoClock.system();

        mode = Mode.IDLE;

        turnController = new PIDFController(HEADING_PID);
        turnController.setInputBounds(0, 2 * Math.PI);

        constraints = new MecanumConstraints(
                BASE_CONSTRAINTS,
                TRACK_WIDTH,
                WHEEL_BASE,
                LATERAL_MULTIPLIER);

        follower = new HolonomicPIDVAFollower(TRANSLATIONAL_PID, TRANSLATIONAL_PID, HEADING_PID);
    }

rbrott, 2020-01-28:
> I modified SampleMecanumDriveBase in road-runner-quickstart.

I should have been more specific. I'm more interested in your modified `MecanumDrive` (i.e, the direct child of `Drive`) and the new kinematics. In particular, I'd like to see `setDriveSignal()`. 

> Also, turning is not properly working. Falls well below target value. I tested with target of 90 degrees but actual was somewhere around 30 degrees.

Is this new? If so, it may have to do with adding in the wheelbase. If you're using the empirical track width as suggested in the quickstart, you should set the wheelbase equal to the track width (this is the default behavior).

> Lastly, I had to create a new motor type for NeveRest 20 Orbital.

In retrospect, the numerous inaccuracies in the built-in motor definitions nullifies their convenience. The next major set of quickstart changes will remove these definitions from `DriveConstants`.

dvpatel, 2020-01-30:
Hi Ryan,

It was a user error.  :-(  I'm all set.  Strafing, turning, all working beautifully.  Thanks for your help!  You've done a great job with this toolkit.

I did have one question around starting position.  I'm not starting at origin.  At starting position, the robot is turned -180 degrees and placed at some coordinate other than origin, ie. -48.0, +48.0.  How do I start at this point?  Any example?  

With heading set at -180 degrees, the robot actually turns to this position before following the trajectory.  I don't want the robot to do this.

-Dipesh

rbrott, 2020-01-31:
You can call `drive.setPoseEstimate()` at the beginning of your op mode to set a different starting pose.

closed 2020-02-02 by dvpatel