acmerobotics/road-runner #31: How to use Heading Interpolators?

LetianLi, 2020-01-24:
I've been looking at the heading interpolators that can be passed into lineTo and splineTo. I cannot tell what they do nor how to use them in the program. 

I believe lineTo is similar to strafeTo, but you can control the heading through the interpolator. Please correct me if im wrong.

LetianLi, 2020-01-24:
what are the differences between ConstantInterpolator, LinearInterpolator, SplineInterpolator, TangentInterpolator, and WiggleInterpolator. 
TangentInterpolator seems to be the default. ConstantInterpolator does not seem useful to my case as it is for holonomic drives.

LetianLi, 2020-01-24:
If there is a documentation i can be pointed to, that would help

rbrott, 2020-01-24:
If you have a nonholonomic drive, you must use the tangent interpolator (`tan(theta) = y'/x'` is the nonholonomic constraint). The other interpolators are for changing the heading of the robot independent of the translational motion.

LetianLi, 2020-01-24:
If i wanted to strafeTo a position in a straight line, but end up at a specific heading, how do i do it

rbrott, 2020-01-24:
Do you have a holonomic drivetrain?

LetianLi, 2020-01-24:
mecanum

rbrott, 2020-01-24:
OK so you do in fact have a holonomic drivetrain. In that case, you can use `LinearInterpolator` or `SplineInterpolator` to smoothly transition from one heading to another across a path segment. If you only have one segment in your path, I recommend `LinearInterpolator` (otherwise use `SplineInterpolator`). As you observed, `strafeTo()` uses `ConstantInterpolator`, while `lineTo()` is more flexible:
```java
Trajectory traj = new TrajectoryBuilder(...)
    .lineTo(new Vector2d(...), new LinearInterpolator(startHeading, angle));
    .build();
```

LetianLi, 2020-01-24:
what goes into variables
`startHeading, angle` 

and what does splineInterpolator do differently compared to linearInterpolator?

LetianLi, 2020-01-24:
is it more like `startingAngle, endingAngle`

LetianLi, 2020-01-24:
looking at the code, `angle` seems to be the amount in radians that the robot turns, not the ending angle. I guess this makes sense. I think I understand heading interpolators now.
However, there are many constructors for splineInterpolator and i do not know what the second derivatives are for start heading and end heading.

LetianLi, 2020-01-24:
it also looks like it will spline the heading "correctly" even when leaving the derivs null. 
Thank you for your help.

closed 2020-01-24 by LetianLi

LetianLi, 2020-01-27:
The derivs aren't even in parameters. But just a question, in what condition would someone need to use the WiggleInterpolator

rbrott, 2020-01-27:
It's useful for intaking/foraging motions.

LetianLi, 2020-01-27:
That makes sense, thank you