acmerobotics/road-runner #32: Dashboard not working on 0.4.7 new install

FTC9977, 2020-01-28:
Sorry for the newbie question here, but our team installed the roadrunner SDK following the instructions provided on GitHub, without any errors.

Problem we have is, when we install the DriveVelocityPIDTuner.java file to the Robot controller, we are unable to web to the dash board, after AS indicates a successful installation was done.

Web Address were trying (Per the comments section of  DriveVelocityPIDTuner.java)
		http:/192.168.49.1:8080/dash

I have verified the PC is connected to the Direct WiFi, as we can pull up the Robot Controller Connection Information, and under active connections, we see our PC listed.   I also checked in the External Libraries and see:

Gradle: com.acmerobotics.dashboard:0.3.6@aar
Gradle: com.acmerobotics.roadrunner:core:0.4.7@jar

Can someone provide some guidance for our team as to what we are missing here?  

Thanks

FTC9977, 2020-01-28:
Also, the message being displayed by the Web Browser is as follows:

Mime type unknown: uri='/dash' path='dash'

rbrott, 2020-01-28:
Did you follow [these instructions](https://acmerobotics.github.io/ftc-dashboard/gettingstarted) for installing the dashboard? Also, can you send a screenshot of the RC activity? Is there any blue text in the main view?

FTC9977, 2020-01-28:
Ryan:

Thank you for responding so quickly...  I indeed failed to apply these
steps, and will be doing so here shortly.    Thank you very much for your
assistance.  I will let you know if that worked.

Kind Regards,
Eric Harwood
Coach - FTC9977

FTC9977, 2020-01-28:
Ryan:
Just a quick note again to say thank you.  I indeed missed those steps, and
now I am seeing the dashboard.

Our team is very grateful for your assistance...

Kind Regards,
Eric Harwood
Coach - #9977 Circuit Shifters

closed 2020-03-14 by rbrott