acmerobotics/road-runner #33: Best method for 1D motion profiling?

nchowder, 2020-01-31:
Let's say I have two motors powering a lift and I want them to move from one position to another with constraints on max accel and velocity. Right now I am just using a PIDF controller tuned for the lift and changing the target position when I want to move the lift, but this can be slow when moving short distances. I know there isn't a built-in drive class for this situation, but what would be the best way to accomplish this goal?

rbrott, 2020-01-31:
For each motion, you should call `MotionProfileGenerator.generateSimpleMotionProfile()` to create a profile. To execute this profile, you can use two separate `PIDFControllers` with proper feedforward (i.e., `kV`, `kA`, `kStatic`). At each time step, use the elapsed time to compute the target kinematic state (position, velocity, acceleration). Set `targetPosition` to the desired position and pass the measured position, desired velocity, and desired acceleration, respectively to `update()`.

nchowder, 2020-02-03:
Is this what you mean?

    PIDFController controller = new PIDFController(...);
    MotionProfile profile = MotionProfileGenerator.generateSimpleMotionProfile(...);
    ElapsedTime timer = new ElapsedTime();
    while (...) {
        MotionState desiredState = profile.get(timer.seconds());
        controller.setTargetPosition(desiredState.getX());
        correction = controller.update(motor.getCurrentPosition(), desiredMotionState.getV(), desiredMotionState.getA());
        motor.setPower(correction);
    }

rbrott, 2020-02-03:
Yes, that looks correct.

closed 2020-02-10 by nchowder