acmerobotics/road-runner #34: [Q] Generate javadoc

Calvin-Xu, 2020-02-03:
I have been trying to generate the javadoc for this project for local reference, but not sure how it was done for a Kotlin project. Would you provide any instruction? A pre-generated archive would be great too.

rbrott, 2020-02-04:
The documentation is written in the KDoc format and processed with Dokka. You can build it by running `./gradlew dokka`. The generated files will be placed in `core/build/kdoc`.

Here's an archive of the 0.4.7 docs: [kdoc-0.4.7.zip](https://github.com/acmerobotics/road-runner/files/4151190/kdoc-0.4.7.zip)

Calvin-Xu, 2020-02-04:
Thanks. So we see that there is also a [Javadoc](https://acmerobotics.github.io/road-runner/core/0.4.6/javadoc/). Since we are still on Java that would be more helpful.

closed 2020-03-14 by rbrott