acmerobotics/road-runner #36: reg MOTOR_VELO_PID Tuning

mohanskumar, 2020-02-06:
Hi there, 

I am rookie for roadrunner and trying to setup for a 5202 435 RPM motors with inbuilt encoders. Just following the instructions to tune MOTOR_VELO_PID. while I am in the Dashboard , I see some values like  0, 3, 10 already loaded. Trying to understand the tuning process itself. It would be great if I could get a guide ( preferably with an example ) on that. Sincerely appreciate the support in this regard.  Not sure how to proceed with tuning process itself.

rbrott, 2020-02-06:
Here's the advice I've been giving for turning the velocity PID:
1. Increase kP until any phase lag is eliminated. Concurrently increase kD as necessary to mitigate oscillations.
2. Add kI (or adjust kF) until the steady state/constant velocity plateaus are reached.
3. Back off kP and kD a little until the response is less oscillatory (but without lag).

Consistency will come with tuning. If you only care about the final position (i.e., the transient trajectory tracking is satisfactory), you can set a non-zero follower timeout.

mohanskumar, 2020-02-07:
Thanks Ryan for the quick response. Sorry that I still could not get the process. few questions thought

1) I see some default values ( 10, 3 , 0 ) in my case , showing up for VELO_PID. where is this coming from? 
2) Increase kP and kD meaning that it should be bumped up gradually ( by +1  or so) ?
3) backoff kP and kD a little ( meaning reducing some decimals ?)
 4) where could I set this non-zero follower timeout ?

rbrott, 2020-02-07:
> I see some default values ( 10, 3 , 0 ) in my case , showing up for VELO_PID. where is this coming from?

Firmware defaults

> Increase kP and kD meaning that it should be bumped up gradually ( by +1 or so) ?

Sure that's reasonable. I usually go a bit more (+2/3), but that's personal preference.

> backoff kP and kD a little ( meaning reducing some decimals ?)

Yes or perhaps more. This really depends on whether the additional integral feedback made things too unstable.

> where could I set this non-zero follower timeout ?

The constructor of your `TrajectoryFollower` subclass. For example, [this line](https://github.com/acmerobotics/road-runner-quickstart/blob/fee96d2f5ecb820083bbb1ba5ca1f835d70074fd/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/drive/mecanum/SampleMecanumDriveBase.java#L79) in the quickstart.

mohanskumar, 2020-02-07:
Thank you so much Ryan. Should I start with Firmware defaults or 0s?

rbrott, 2020-02-07:
Firmware defaults is fine. Doubtful you'll want to go any lower.

closed 2020-02-07 by mohanskumar

mohanskumar, 2020-02-07:
cool. Sincerely appreciate your prompt response.