acmerobotics/road-runner #37: Trajectory Caching?

ToiletCommander, 2020-02-18:
Is there a possible way to cache generated trajectories in a file?

jeevan-navudu, 2020-02-19:
How are you generating these trajectories? For example, are you generating these trajectories using road-runner-plugin? If you are using a road-runner-plugin, you could save the trajectories to a file and load them later.

rbrott, 2020-02-19:
No, there is built-in mechanism for doing that. As the other commenter mentioned, you can save trajectory configurations to disk and regenerate them on demand (although this still requires generation). The generation routine is designed to be fast enough to run at the start of every op mode.

ToiletCommander, 2020-02-20:
Thank you for your comments!

closed 2020-02-20 by ToiletCommander