acmerobotics/road-runner #38: Tangent Field of GUI meaning and usage

Superman132, 2020-02-22:
Hi from Team 11172 Static Discharge,

I wanted to first say how much I admire and appreciate your work on this and other FTC libraries. Thank you! I wanted to ask what the purpose of the tangent field in the GUI means, and whether I have to worry about it in my programmed trajectory. I looked at the heading Interpolators and they don't seem to mention anything about a tangent angle in the constructor. I deduced that it is the angle of entry to the final path position. Is that correct, or is there a gap in my understanding?

Thanks

rbrott, 2020-02-23:
> I wanted to ask what the purpose of the tangent field in the GUI means, and whether I have to worry about it in my programmed trajectory.
> I deduced that it is the angle of entry to the final path position.

Yes, that's correct. The unit tangent vector is given by the first derivative of the arclength parametrization (i.e., the unit-length derivative vector for a smooth parametrization).

Superman132, 2020-02-23:
If I am not generating splines using waypoints and deriatives, but just using the Trajectory Builder, can/should I worry about this value. It seems there is no option in the TrajectoryBuilder to make splines with specific tangent values.

rbrott, 2020-02-23:
Yes, you should worry about this value. While this isn't super clear from the docs, the `heading` field of the `Pose2d` object passed to `splineTo()` is actually used to determine the tangent direction (i.e., the angle that the path enters the waypoint). When using the tangent interpolator, this coincides with the robot heading at the waypoint. The tangent vector magnitude is automatically selected based on the path length.

Superman132, 2020-02-23:
Okay, I think that is what was confusing me. I had been trying to get a good understanding of how roadrunner works, and I thought the Pose2d of the localizer, which i believed\is (x, y robot heading) was the same as the splines pose2d. But the code wasn't lining up with this belief. Also, I wondered why the splineTo function had a heading parameter, because I thought it could just be inferred from the pose2d. But now it all clicks together. If its possible, could I maybe make a troubleshooting doc page about this to help newcomers?

rbrott, 2020-02-23:
Yes, the docs will be updated when 0.5.0 is released. Unfortunately `Pose2d` has many hats.

closed 2020-02-24 by Superman132