acmerobotics/road-runner #42: 0.5.0 PathContinuityViolationException

shishghate, 2020-02-27:
We are using the visualizer and running the following code:

        val builder1 = TrajectoryBuilder(Pose2d(-40.0, -63.0), false, constraints)

        builder1
            .strafeTo(Vector2d(-20.0, -32.0))// pick up first block
            .splineTo(Pose2d(0.0, -36.0))

We get an exception when processing the splineTo.  And suggestions as to what we should do differently?

Superman132, 2020-02-28:
I'm not the author/maintainer of this library, but I take a shot at answering this. I believe the problem lies in a mismatch of heading interpolators between `strafeTo` and `splineTo`. The heading interpolator for `strafeTo` is constant, because it holds the robot angle constant and only moves sideways. Meanwhile the heading interpolator for `splineTo` is Tangent Interpolator, which requires that the robot stay tangent to the direction of its path. The problem is that you can't use a Tangent Interpolator with any other interpolator because it has that specific requirement which invalids certain robot and path configurations that are allowed on other Interpolator. Good News, The fix is just to change `splineTo` to something like `splineToConstantHeading()` for a constant heading, `splineToLinearHeading()` for a linear heading interpolator to a certain angle, spline interpolator is similar. This gives you extra flexibility that doesn't come with the default spline Tangent Interpolator. The maintainer can correct me if this explanation is wrong/misleading, but it seems to me to be the issue.

rbrott, 2020-02-28:
The issue here is a bit subtle. The builder changes in 0.5.0 were deliberately designed to expose latent issues like this with existing trajectories. The starting tangent direction (i.e., 0) does not match the tangent angle of the line. Once you fix this, you will also have to specify 0 as the start heading (start heading defaults to the tangent orientation). Here's a modified snippet:

```kotlin
val traj = TrajectoryBuilder(Pose2d(-40.0, -63.0, Vector2d(20.0, 31.0).angle()), 0.0, constraints)
    .strafeTo(Vector2d(-20.0, -32.0))// pick up first block
    .splineTo(Pose2d(0.0, -36.0))
    .build()
```

shishghate, 2020-02-28:
Ok thanks, that makes sense.  And we modified our code to use the splineToConstantHeading.  But we are running into a small issue with the reverse path.   When we run this sample code in the visualizer, we are seeing that small forward/reverse movement at the beginning/end of each segment.  Here is our code:

        var builder2 = TrajectoryBuilder(Pose2d(54.0, -30.0), false, constraints);

        builder2
            .splineToConstantHeading(Pose2d(0.0, -36.0))
            .splineToConstantHeading(Pose2d(-44.0, -32.5)) // pick up second block

And here is a visual:

[image]

Any help would be appreciated and thanks.

shishghate, 2020-02-28:
Nevermind, we just figured it out:

        var builder2 = TrajectoryBuilder(Pose2d(54.0, -30.0, 180.0.toRadians), true, constraints);

        builder2
            .splineToConstantHeading(Pose2d(0.0, -36.0, 180.0.toRadians))
            .splineToConstantHeading(Pose2d(-44.0, -32.5, 180.0.toRadians)) // pick up second block

closed 2020-03-02 by rbrott