acmerobotics/road-runner #43: Question involving how trajectories are calculated

LetianLi, 2020-02-29:
I am thinking of joining Roadrunner's Path generation with Pure Pursuit's following system. Pure pursuit uses a circle to intersect with a path so I want to know if Roadrunner generates the path as a series of equations, a series of points, or whatever else. If so, how can i extract these equations. Also, is it even possible to calculate an intersection? (side question, does the dashboard receive individual points to draw or the equation of a line, etc.)

rbrott, 2020-02-29:
> I want to know if Roadrunner generates the path as a series of equations, a series of points, or whatever else.

Road Runner represents paths with parametric curves (i.e., an explicit, displacement-parametrized equation for each pose component). From this representation, you can retrieve the pose corresponding to a given displacement as well as the first and second derivatives of the pose. 

> Also, is it even possible to calculate an intersection?

No, there is no closed-form expression for circle-curve intersection in general. You may want to sample points from the path and construct a linear approximation.

> does the dashboard receive individual points to draw or the equation of a line, etc.

You can see all of the dashboard canvas primitives here: <https://github.com/acmerobotics/ftc-dashboard/blob/master/FtcDashboard/src/main/java/com/acmerobotics/dashboard/canvas/Canvas.java>

LetianLi, 2020-02-29:
Thanks for your response. What do each of these derivatives mean?

LetianLi, 2020-02-29:
I'm looking at the [ParametricCurve.kt](https://github.com/acmerobotics/road-runner/blob/master/core/src/main/kotlin/com/acmerobotics/roadrunner/path/ParametricCurve.kt) file

rbrott, 2020-02-29:
I suggest you read [this doc](https://github.com/acmerobotics/road-runner/blob/master/doc/pdf/Quintic_Splines_for_FTC.pdf). The code in `ParametricCurve.kt` is a fairly direct implementation of that math.

LetianLi, 2020-02-29:
It seems that the curve is constructed with many vectors. I can work with vectors (linear lines) with pure pursuit. However, do these vectors have a bounding box (domain and range) to ensure that they're not an infinite line?

rbrott, 2020-02-29:
> do these vectors have a bounding box (domain and range) to ensure that they're not an infinite line?

I'm not sure what you're asking. Every position on a built-in parametric curves has finite components.

LetianLi, 2020-02-29:
Pure Pursuit normally uses several line segments that are each connected to one another. This is the path. The follow point is one of the intersections between the path and the circle. Because intersection is mathematically using the equations of the lines, it may return an intersection with the circle off of the path. To solve this, I'm checking to see if the intersection is within the minX, maxX, and minY, maxY of the line.

I'm asking if the vector has an end on both sides.

rbrott, 2020-02-29:
> Because intersection is mathematically using the equations of the lines, it may return an intersection with the circle off of the path. To solve this, I'm checking to see if the intersection is within the minX, maxX, and minY, maxY of the line.

That check doesn't sound very robust. It's better to imagine the line segment as the image of a parametric function of a single bounded parameter and compute intersections in terms of that parameter. Then this check becomes the simple matter of checking whether the intersection parameter value lies within the aforementioned bounds. For instance, one possible representation line segment connecting the points **x**_0 = (x_0, y_0) and **x**_1 = (x_1, y_1) is the parametric curve **x**(t) = **x**_0 + (**x**_1 - **x**_0) t where t lies in the interval [0, 1].

LetianLi, 2020-02-29:
This is what I currently have. (x1, y1) being intersection 1, (x2, y2) being intersection 2. p1 and p2 are the line vertices. 
```
boolean validIntersection1 = Math.min(p1.x, p2.x) < x1 && x1 < Math.max(p1.x, p2.x)
                    || Math.min(p1.y, p2.y) < y1 && y1 < Math.max(p1.y, p2.y);
boolean validIntersection2 = Math.min(p1.x, p2.x) < x2 && x2 < Math.max(p1.x, p2.x)
                    || Math.min(p1.y, p2.y) < y2 && y2 < Math.max(p1.y, p2.y);
```

LetianLi, 2020-02-29:
Do I use the f(t)=p1+(p2-p1)t, to calculate intersection, or to check validity, or both?

How would I get the vectors in the parametric curve and how is it returned? (looks like there's a get(t, s); in the ParametricCurve.kt)

rbrott, 2020-02-29:
> Do I use the f(t)=p1+(p2-p1)t, to calculate intersection, or to check validity, or both?

I would setup your equations to solve for t and then you basically get the validity check for free. You can also use your method, but you should be careful of the vertical/horizontal degenerate cases.

> How would I get the vectors in the parametric curve and how is it returned? (looks like there's a get(t, s); in the ParametricCurve.kt)

The parameter "s" is the arc length displacement along the curve that assumes values in [0, length]. The parameter "t" is some internal parameter that assumes values in [0, 1]. The return values are of type `Vector2d`.

LetianLi, 2020-02-29:
Isn't Vector2d a point?

rbrott, 2020-02-29:
It can be. It can also be a first derivative or a second derivative. It's really a generic container for (x, y) paired data.

LetianLi, 2020-02-29:
So what two numbers are stored in this Vector2d?

rbrott, 2020-03-01:
The `get()` return value is indeed a position.

LetianLi, 2020-03-02:
For my purpose, do i try to do something like this?
```
for (var i = 0; i < curve.length(); i++) {
  checkIntersection(curve.get(i, 0), curve.get(i, 1));
}
```
I think what this is doing is checking the vector between those 2 points at i of the arc length displacement.
Would this work?

rbrott, 2020-03-02:
> I think what this is doing is checking the vector between those 2 points at i of the arc length displacement.

There are a couple issues:
1. Displacement along the path is a `double`-valued quantity. You'll have to choose a suitable discretization for your application (which may or may not be one inch).
1. You should omit the second parameter to `get()`. The only reason to pass `t` explicitly is to implement performance optimizations.

Beyond that, I obviously can't comment on the correctness of `checkIntersection()`.

LetianLi, 2020-03-02:
I have two parameters of checkIntersection() which are the two points of the line i am checking. 
Would this be correct?
```
double resolution = 0.25; // is this inches? Am i sampling points every 0.25 inches?
for (var i = 0; i * resolution < curve.length() - resolution; i++) {
  Vector2d p1 = curve.get(i * resolution);
  Vector2d p2 = curve.get((i + 1) * resolution);
  checkIntersection(p1, p2);
}
```

rbrott, 2020-03-02:
Sure

LetianLi, 2020-03-02:
How do i build a trajectory without the wheel velocities and time stuff so that i can access the parametric curves?

LetianLi, 2020-03-02:
is there a difference between SimpleTrajectoryBuilder and TrajectoryBuilder?

LetianLi, 2020-03-02:
Also, would the last point be `curve.get(curve.length())` or `curve.get(curve.length(), 1)`

rbrott, 2020-03-02:
> How do i build a trajectory without the wheel velocities and time stuff so that i can access the parametric curves?

The highest-level path API is `PathBuilder`.

> is there a difference between SimpleTrajectoryBuilder and TrajectoryBuilder?
Edit: There's also a BaseTrajectoryBuilder?
Edit: as well as TrajectoryGenerator? What's the difference between all of them

`TrajectoryGenerator` -- low-level interface for generating trajectories of all kinds
`BaseTrajectoryBuilder` -- base trajectory builder class that delegates primarily to an internal `PathBuilder`
`TrajectoryBuilder` -- specific trajectory builder for trajectories **with dynamic/displacement-varying velocity and acceleration constraints**
`SimpleTrajectoryBuilder` -- specific trajectory builder for trajectories **with static/displacement-invariant velocity, acceleration, and jerk constraints**

> Also, would the last point be `curve.get(curve.length())` or `curve.get(curve.length(), 1)`

Those are both correct, although the preferred method for retrieving the final curve position is `end()`.

LetianLi, 2020-03-04:
`PathBuilder` will create a `Path` Object which contains `PathSegment`s which are basically a `ParametricCurve`+`HeadingInterpolator`. I'm currently using the `get()` inside the `Path `so I think that's fine.

I also found a `PathFollower`. How does `PathFollower` differ from `TrajectoryFollower`? And how does it create a `DriveSignal`?

rbrott, 2020-03-05:
> I can use the `Path`'s `get()` right?

Yes

> Do I need to write an override addon class to `PathFollower`similar to how there's a `HolonomicPIDVAFollower` for `TrajectoryFollower`? 

Yes, you should make a `PathFollower` subclass.

> Does it also need to be in Kotlin?

No

LetianLi, 2020-03-05:
PP works by chasing a point, and I have that point. Since it's not time based, how should I fill in the `targetRobotVel` and `targetRobotAccel` for the `DriveSignal`.

LetianLi, 2020-03-05:
You can see [here](https://github.com/LetianLi/ftc-skystone-from-road-runner/blob/master/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/yoda_code/PurePursuitFollower.kt). I copied from the HolonomicPIDVAFollower and then made adaptations.

rbrott, 2020-03-05:
> PP works by chasing a point, and I have that point. Since it's not time based, how should I fill in the `targetRobotVel` and `targetRobotAccel` for the `DriveSignal`.

The ultimate purpose of the follower is to provide a robot velocity/acceleration. For many followers, that process may involve computing the desired pose and closing the loop based on the measured/localizer pose. This is true for pure pursuit; once the lookahead point has been found, the algorithm computes the angular velocity for the chosen longitudinal/forward velocity based on an arc connecting the measured position to the lookahead position. You are free to choose the forward velocity however you like---you can keep it constant or do some basic online motion profiling. You can leave the acceleration zero for now.

LetianLi, 2020-03-05:
Both of these values need to be Pose2d. So I'm not sure what goes into the constructors for these values.
Also, would those values be set, or do I need to compute for them to change manually. In other words, is it built into Kinematics and PathFollower.

rbrott, 2020-03-06:
> Both of these values need to be Pose2d. So I'm not sure what goes into the constructors for these values.

`targetRobotVel = new Pose2d(forwardRobotVel, 0, angularRobotVel)`

>  Also, would those values be set, or do I need to compute for them to change manually. In other words, is it built into Kinematics and PathFollower.

You need to compute them fresh on each iteration. That's part of the pure pursuit math.

LetianLi, 2020-03-06:
What units are these? for vel and accel.

rbrott, 2020-03-06:
(path distance units)/s, rad/s and similarly for acceleration with 1/s^2

LetianLi, 2020-03-06:
is the `targetVel` field based or robot based.

LetianLi, 2020-03-06:
I think it's field, ok

LetianLi, 2020-03-06:
Can you just check, something doesn't seem right here. I'm converting between 
```
val targetPose = getFollowPoint(path, currentPose, searchRadius)

val absoluteXToPoint = targetPose.x - currentPose.x
val absoluteYToPoint = targetPose.y - currentPose.y

val absoluteAngleToTarget = atan2(absoluteYToPoint, absoluteXToPoint)
val relativeAngleToPoint = radWrap(absoluteAngleToTarget - currentPose.heading)

val relativeTurnAngle = radWrap(relativeAngleToPoint + targetPose.heading)
val movement_turn = Range.clip(relativeTurnAngle / Math.toRadians(30.0), -1.0, 1.0) * turnSpeed

val targetVel = Pose2d(absoluteXToPoint, absoluteYToPoint, movement_turn) * constraints.maxVel
val targetAccel = Pose2d() * constraints.maxAccel

val targetRobotVel = Kinematics.fieldToRobotVelocity(targetPose, targetVel)
val targetRobotAccel = Kinematics.fieldToRobotAcceleration(targetPose, targetVel, targetAccel)

return DriveSignal(targetRobotVel, targetRobotAccel)
```

rbrott, 2020-03-06:
`DriveSignals` are always specified in the robot coordinate frame.

LetianLi, 2020-03-06:
but things before the fieldToRobot...() are in the global frame right?

rbrott, 2020-03-06:
Yes, `Trajectories` always return quantities in the field coordinate frame.

LetianLi, 2020-03-09:
What are the Pose Derivatives that get returned in `Path.kt`? (eg velo, accel, jerk for trajectory)

rbrott, 2020-03-09:
They don't really have specific names except perhaps the curvature ||**r**''(s)||. They are simply pose derivatives with respect to displacement.

LetianLi, 2020-03-09:
Tangent Line?

rbrott, 2020-03-09:
Well you can parametrize the tangent line by **v**(s) = **r**(s_0) + **r**'(s_0) s. 

I suppose you can think of the derivatives in terms of Frenet frames.

LetianLi, 2020-03-10:
v(s) = r(s_0) + r'(s_0) s
Where:
v(s)    : function of the tangent line, returning a point
s        : displacement on the tangent line
r(s_0) : function of the path
r'(s_0): derivative of r(s_0)
s_0    : displacement on the path

closed 2020-05-16 by rbrott