acmerobotics/road-runner #44: Forward, Reverse with tank drive

EddieDL, 2020-03-05:
Do the latest changes changes mean you can't make a trajectory that goes from moving forward to reverse on a tank drive?

I am looking to make a motion similar to the following:
```
tb.lineTo(new Vector2d(50,0));
tb.splineTo(new Pose2d(0, 100, Math.toRadians(-90)));
```
- Move forward 50 inches
- Move backward in a curving motion to the point 0,100 inches.

Is the only way to do this now to have two trajectories?

rbrott, 2020-03-05:
> Is the only way to do this now to have two trajectories?

Yes. 0.5.0 `TrajectoryBuilder`/`PathBuilder` actually enforces continuity.

EddieDL, 2020-03-05:
So to make this motion would need to use code similar to the following. Obviously we would need to follow each of these Trajectories in order.
```
TrajectoryBuilder tb = new TrajectoryBuilder(Pose2d(0,0,0), constraints);
tb.lineTo(new Vector2d(50,0));
tb.build();
TrajectoryBuilder tb2 = new TrajectoryBuilder(Pose2d(50,0,0), true, constraints);
tb2.splineTo(new Pose2d(0,100, Math.toRadians(-90)));
tb2.build();
```

Also, will the `splineTo()` method work to create a curve with a tank drive? I assume so, but the motion I just saw didn't seem to match.

rbrott, 2020-03-05:
> Also, will the `splineTo()` method work to create a curve with a tank drive? I assume so, but the motion I just saw didn't seem to match.

Yes, `splineTo()` works fine for tank drives. There is a subtle issue with the trajectories you presented. The angle you pass in `Pose2d` represents the tangent to the path. Normally, this is the same as the heading; however, for reversed paths, the tangent and heading point in opposite directions. To ensure continuity, the headings must match, while the tangents do not have to match. In the snippet you posted, the first trajectory ends with heading 0, and the second trajectory starts with heading 180 (i.e., the opposite of the tangent, 0). To remedy this, simply change the start tangent of the second trajectory to 180:
```java
TrajectoryBuilder tb = new TrajectoryBuilder(Pose2d(0,0,0), constraints);
tb.lineTo(new Vector2d(50,0));
tb.build();
TrajectoryBuilder tb2 = new TrajectoryBuilder(Pose2d(50,0,Math.toRadians(180)), true, constraints);
tb2.splineTo(new Pose2d(0,100, Math.toRadians(-90)));
tb2.build();
```

EddieDL, 2020-03-05:
Ahh... That makes sense. I am pretty sure I read another issue where someone responded very similarly, but I couldn't quite put it together. Now I get it.

Given reverse trajectories use the opposite heading, can I assume the same could be said of the resulting ``Pose2d` objects returned by ``Trajectory.get(time)``? For example the the first Trajectory the heading will be 0, but the second would start at 180?

Related question, my target in the ``splineTo()`` is -90. How is this impacted by the fact that the starting `Pose2d` is 180?

rbrott, 2020-03-05:
`Trajectory.get()` always returns the true robot heading (which is not necessarily the tangent). 

> For example the the first Trajectory the heading will be 0, but the second would start at 180?

No, the heading is zero in both cases.

> Related question, my target in the `splineTo()` is -90. How is this impacted by the fact that the starting `Pose2d` is 180?

It isn't.

closed 2020-03-06 by EddieDL