acmerobotics/road-runner #46: How do you get the robot heading and angular velocity at a point in time in a Trajectory?

EddieDL, 2020-03-06:
So I am trying to use road-runner to generate Trajectories. Then I want to pull the following data for each 10ms of the Trajectory and pump this data into a motor controller which will actually control robot motion.
- Position
- Velocity
- HeadingAngle
- HeadingAngularVelocity

I currently use `Trajectory.get(n)` and `Trajectory.get(n-1)` and then calculate these values. Which seems to work fine for Position and Velocity, but what I am doing for the heading isn't working at all. Currently I am doing `Trajectory.get(n).getHeading()` for the HeadingAngle and `Trajectory.velocity(n).getHeading()` for HeadingAngularVelocity.

After looking at things I think I might be able to use `Trajectory.getMotionProfile().get(n)` for some of this. But I am not 100% sure what `MotionState` properties are.

Any suggestions or thoughts?

EddieDL, 2020-03-06:
Just to add a little bit more clarity, I am trying to make the following motion using two Trajectories. One is forward, the other is reverse.

[image]

rbrott, 2020-03-06:
You just need to alter the second trajectory like so:
[image]

EddieDL, 2020-03-06:
Thanks! 

This seems to imply that when the "front" of the robot is pointing right the heading is positive. Can I also assume that if the "front" is pointing left the heading is negative?

As you can probably guess I am not using road-runner for an FTC game. Is the GUI tool able to be customized with a different image and different "size"?

What does resolution do? I have seen it in some of the code and it is all defaulted to .25, but it isn't clear to me what it means.

rbrott, 2020-03-06:
> This seems to imply that when the "front" of the robot is pointing right the heading is positive. Can I also assume that if the "front" is pointing left the heading is negative?

Yes, that's true. Heading is measured counterclockwise from the x-axis. If you provide negative/out-of-range values, they will be normalized.

> As you can probably guess I am not using road-runner for an FTC game. Is the GUI tool able to be customized with a different image and different "size"?

You'll have to modify the source to use a different image. If you give me an image, I can probably build you a jar with that image.

> What does resolution do? I have seen it in some of the code and it is all defaulted to .25, but it isn't clear to me what it means.

The resolution is used for sampling the constraint functions to generate the motion profile. The smaller the resolution, the better the functional constraints are obeyed, but the longer the generation takes.

EddieDL, 2020-03-06:
This all makes sense.

I might take you up on the offer of updating the GUI tool. I think using that will help resolve a number of my current issues.

You never really answered the crux of my question. My assumption is that means what I am doing now is correct?

> I currently use Trajectory.get(n) and Trajectory.get(n-1) and then calculate these values. Which seems to work fine for Position and Velocity, but what I am doing for the heading isn't working at [...]

>After looking at things I think I might be able to use Trajectory.getMotionProfile().get(n) for some of this. But I am not 100% sure what MotionState properties are.

rbrott, 2020-03-08:
> You never directly answered the crux of my question. My assumption is that means what I am doing now is correct?
> 
> > I currently use Trajectory.get(n) and Trajectory.get(n-1) and then calculate these values. Which seems to work fine for Position and Velocity, but what I am doing for the heading isn't working at all. Currently I am doing Trajectory.get(n).getHeading() for the HeadingAngle and Trajectory.velocity(n).getHeading() for HeadingAngularVelocity.
> 
> > After looking at things I think I might be able to use Trajectory.getMotionProfile().get(n) for some of this. But I am not 100% sure what MotionState properties are.

The same approach you used for translation should work for angular motion. Is it perhaps a radians/degrees issue? There's no need to look at the motion profile.

EddieDL, 2020-03-08:
Ok I think I know what the problem is. When I get the "distance" I need the distance along the curve not the straight line distance between the two points identified by the pose. Is there any way to get that distance from the trajectory?

rbrott, 2020-03-08:
In that case, you want to use the underlying motion profile.

EddieDL, 2020-03-08:
How can I use the MotionProfile? I know how to get it, but I don't understand what A, J, V and X are on the ProfileState.

rbrott, 2020-03-09:
A, J, V, and X are acceleration, jerk, velocity, and position, in that order. In the context of trajectory generation, the position in question is the displacement along the path (it sounds like that's what you want).

EddieDL, 2020-03-09:
So I think I have the solution to my issue. The solution I am using uses an average position value that is taken between the right and left encoder counts. To support this, I calculated two different segment lengths and averaged them. I was hoping to get that from road-runner but I don't think that is how any of the followers work.

I haven't tried it out yet (that will be for tonight), but I think it will work.

EddieDL, 2020-03-10:
So I am trying to use the GUI to check that I am doing things correctly and I keep getting "empty path".

[image]

rbrott, 2020-03-10:
The builder can't create a path with only a single pose. You can add more poses/knots with the button on the right.

NoahBres, 2020-03-11:
@EddieDL I highly recommend joining the FTC Discord or asking these questions on Reddit. Many of these issues have already been explored and there's a ton of awesome people (especially on the Discord) that are willing to quickly reply.
GitHub issues isn't the ideal medium for these.

EddieDL, 2020-03-11:
@rbrott Sorry I wasn't clear. Below is exactly what I did.
- Select an empty folder for the "project".
- Adjusted Step 1
- Clicked Add
Result: nothing happened
- Clicked config
- Switched to TANK drive type
Result: status: error: empty path

So I guess my real problem is why I can't click add. I am sure I am not doing something simple, just not sure what it is.

EddieDL, 2020-03-11:
@NoahBres I will check out the FTC Dicsord. Thanks.

rbrott, 2020-03-14:
It seems like that sequence of interactions does get you into a bad state. Before you interact with the path controls, you should add a trajectory in the panel on the left side.

EddieDL, 2020-03-14:
That fixed it, I couldn't figure out which order I needed to do things.

Thanks!!

closed 2020-03-14 by EddieDL