acmerobotics/road-runner #49: Question is there a way to save code made in the plugin

AlessioToniolo, 2020-04-17:
Hi i'm a rookie and I downloaded the plugin into android studio and I was making some paths and was wondering if there is a way to save the code in the plugin for the path and trajectories so I could use that in autonomous. Or does the gui do this not the plugin? I could not get the gui downloaded. Also I have a question on roadrunner. Is roadrunner also like an odometry with motor encoders or is it just for movement? For example if someone bumped into a robot using roadrunner would the motor encoders detect and make a new path back to the original trajectory? Thanks!

rbrott, 2020-04-19:
> I downloaded the plugin into android studio and I was making some paths and was wondering if there is a way to save the code in the plugin for the path and trajectories so I could use that in autonomous. Or does the gui do this not the plugin?

The GUI and plugin both save trajectory files that can be reloaded (they **do not** produce code). To load these files, check out [TrajectoryConfigManager](https://github.com/acmerobotics/road-runner/blob/master/core/src/main/kotlin/com/acmerobotics/roadrunner/trajectory/config/TrajectoryConfigManager.kt). Currently the new file formats and GUI/plugin somewhat in flux. You can expect more stability and convenience in the next few 0.5.x releases.

> Is roadrunner also like an odometry with motor encoders or is it just for movement?

RR has position feedback that closes the loop between the measured position (obtained from motor encoders, separate dead/tracking wheels, or another source). Good feedforward is essential to tracking the trajectories effectively, but it is always paired with feedback.

> For example if someone bumped into a robot using roadrunner would the motor encoders detect and make a new path back to the original trajectory?

Disturbances like bumps are handled by the feedback mechanisms so long as the position estimates are good and there is ample room for velocity/voltage modulation (i.e., there is space between the maximum velocity provided in the trajectory constraints and the true maximum velocity). By default, the built-in followers will not plan a new path/trajectory, but you can accomplish that on the user side with some effort.

AlessioToniolo, 2020-04-20:
Thank you for your response. I have some other questions based on your answer. Can I still get the position with the velocity pid option instead of the feedforward option? Can I get pid with feedforward? I guess I might have feedforward and velocity pid mixed up, not sure. Also, is there a way to access the absolute position from the motor encoders going to the drive motors like as an object or variable? Thanks for all y'all's work!

rbrott, 2020-04-20:
> Can I still get the position with the velocity pid option instead of the feedforward option?

Yes, velocity PID still has velocity feedforward.

> Can I get pid with feedforward? I guess I might have feedforward and velocity pid mixed up, not sure. 

Yes, you can have feedback (e.g., PID) and feedback at the same time.

> Also, is there a way to access the absolute position from the motor encoders going to the drive motors like as an object or variable?

That's below the abstraction level of RR. If you're using the FTC SDK, you can use `DcMotor.getCurrentPosition()` to get encoder positions. The readings are not absolute; the quadrature inputs do not give the controller enough information to determine the absolute position (although RR only cares about relative).

closed 2020-04-20 by AlessioToniolo