acmerobotics/road-runner #51: math in kinematics

wangxdflight, 2020-06-07:
Is there any reference or document to follow for the math here?
https://github.com/acmerobotics/road-runner/blob/master/core/src/main/kotlin/com/acmerobotics/roadrunner/kinematics/Kinematics.kt

It seems it follows math in chapter 3.1 of this document: https://github.com/acmerobotics/road-runner/blob/master/doc/pdf/Mobile_Robot_Kinematics_for_FTC.pdf
Is there any reference for it? This document is too concise to understand.

I understand the math at below link, but cannot correlate it with the link above and the code.
https://ocw.mit.edu/courses/electrical-engineering-and-computer-science/6-186-mobile-autonomous-systems-laboratory-january-iap-2005/study-materials/odomtutorial.pdf

Can you help?
Thanks.

rbrott, 2020-06-07:
The presentation of Chapter 3.1 is borrowed from Section 10.2 of [this book](https://file.tavsys.net/control/controls-engineering-in-frc.pdf). I intentionally omitted the derivation of the formula because I don't think it's particularly insightful. The integration is a good exercise for the reader. 

The external paper you linked takes an alternate, geometric approach. This approach is elegant and intuitive for unicycles, but it doesn't generalize cleanly to robots that can strafe/translate laterally. It relies heavily on arcs which do not completely describe the constant velocity trajectories of strafing vehicles. In fact, to establish that the unicycle trajectories are indeed arcs requires some calculus anyway. I prefer the purely algebraic derivation which is less visual but operates with clear, minimal assumptions. A bit of calculus is required, although that's somewhat inevitable when dealing with things that move.

wangxdflight, 2020-06-08:
Thank you. Very helpful.

closed 2020-06-08 by wangxdflight