acmerobotics/road-runner #52: Trouble with Plugins

jrasor, 2020-06-12:
We (FTC 5197) have the regular RoadRunner and RoadRunner quickstart repos, and they sync in Android Studio 4.0. But trouble with plugins. Maybe it's just a matter of fixing the documentation.

Looking at https://github.com/acmerobotics/road-runner#core. At bottom, procedure for installing the plugin:

> Road Runner also includes a simple IDEA/Android Studio plugin.

What is the name of this plugin?

> Download the latest plugin zip from Releases

There’s no zip in https://github.com/acmerobotics/road-runner/releases. 

> ...or build it with ./gradlew buildPlugin.

We did that, success claimed.
Installation procedure continues:

> In Android Studio, navigate to Settings > Plugins.
> Click the button that reads Install plugin from disk....
> Select the zip archive from earlier.
> Restart Android Studio to activate plugin changes.

Why are we doing this? Didn't ./gradlew buildPlugin just now build it?
Android Studio - Settings - Plugins - I see no new looking plugin. Almost all are bundled. It would help if I knew the name of the plugin just built. 
Restarted Studio, put something in the editor. There is no "Road Runner" on the right side of the editor as promised.

Good luck to all FTC! Get out there and Change the Game!

rbrott, 2020-06-12:
> > Road Runner also includes a simple IDEA/Android Studio plugin.
> 
> What is the name of this plugin?

The name is [Road Runner Path Designer](https://github.com/acmerobotics/road-runner/blob/3b8fc09325fad4529584a45abeb0dc85abfee7ee/plugin/src/main/resources/META-INF/plugin.xml#L2).

> > Download the latest plugin zip from Releases
> 
> There’s no zip in https://github.com/acmerobotics/road-runner/releases.
> 

The archive is located in the assets section at the bottom of every release. The current archive can be found [here](https://github.com/acmerobotics/road-runner/releases/download/v0.5.1/road-runner-plugin-0.5.1.zip).

> We did that, success claimed.
> Installation procedure continues:
>
>
> Why are we doing this? Didn't ./gradlew buildPlugin just now build it?
> Android Studio - Settings - Plugins - I see no new looking [...]

The plugin build creates an archive in `plugin/build/distributions`. You still need to install from disk with this method.

jrasor, 2020-06-12:
Good work, Ryan! 

The archive is located in the assets section ... yup, got it.
The plugin build creates an archive in plugin/build/distributions.  Yup, it's there. On installation, gotta check it in the left list, and restart Studio. I installed it correctly using both methods.

We can close the issue.

closed 2020-06-12 by jrasor