acmerobotics/road-runner #59: Quick Question: How does RoadRunner approxiate path arc length?

Superman132, 2020-11-19:
I was looking at the source code for the RoadRunner paths and the document in the doc folder, and I saw an integral for the arc length of a path using numerical integration. Does the code use a specific numerical integration method for path length?

rbrott, 2020-11-19:
[This post](https://www.chiefdelphi.com/t/real-time-path-planning/350392/13) gives a good overview of the method.

Superman132, 2020-11-20:
I was thinking of writing a pull request with Midpoint Formula numerical integration for the arc length, just for the experience. It shouldn't be too hard because the Spline class always calculates x and y derivatives with respect to t, so I can just integrate that. Would that be a good idea? or would I just be wasting my time?

rbrott, 2020-11-20:
I think that's a great exercise for personal learning, especially if you want to practice/brush up on your calculus, but I'm hesitant to merge anything unless it's significantly superior to the current arc length method.

Superman132, 2020-11-20:
I was attempting to build a fork of the current repo on AS, but it keeps giving me an error. 

**Gradle sync failed: Could not find matching constructor for: org.gradle.api.publish.ivy.internal.publisher.IvyDescriptorFileGenerator(org.gradle.api.publish.ivy.internal.publication.DefaultIvyPublicationIdentity)**

Any ideas on how to fix this?

rbrott, 2020-11-21:
I doubt this repository will open in Android Studio as it doesn't directly depend on Android. I personally use [IDEA](https://www.jetbrains.com/idea/).

Superman132, 2020-11-23:
I did some testing with arc length integration. It mostly failed, even when I experimented with Midpoint, Trapezoidal, and Simpson's rule. Quite disappointing, but a good practice with my calculus skills. Should also be a good indication that the current method is quite optimal :).

closed 2020-11-23 by Superman132