acmerobotics/road-runner #60: Issues With Moving Backwards in Tank Drive

Dinomasater186, 2020-11-24:
When I tell the robot to follow a spline or line backward it turns around and then starts spinning wildly. My best guess is that it is trying to face backwards but it also wants to face forwards to follow the path. Do you know how to fix this?

rbrott, 2020-11-25:
Yes, there's probably with the heading and/or tangent that you've specified. Can you post the problem trajectory?

Dinomasater186, 2020-11-25:
trajectory = drive.trajectoryBuilder(new Pose2d()).splineTo(new Vector2d(30, 30), 0).build();
drive.followTrajectory(trajectory);
SafeSleep.sleep_milliseconds(this, 500);
drive.followTrajectory(drive.trajectoryBuilder(trajectory.end(), true).splineTo(new Vector2d(0, 0), Math.toRadians(180)).build());

Dinomasater186, 2020-11-25:
should it be 0 instead of 180?

rbrott, 2020-11-25:
That looks correct to me for v0.5.1+. Are you using any feedback? If so, I would temporarily turn it off and check what happens with feedforward alone.

Dinomasater186, 2020-11-26:
Thanks for the help. We have fixed this by changing the TankPIDVAFollower so that the cross track controller uses the heading error.

rbrott, 2020-11-27:
Oh I actually helped someone earlier with this problem. I wouldn't recommend switching to heading control; minimizing cross-track error is more important. Here's a patched version of the follower that fixes the sign of the error: https://hastebin.com/neqafeqehi.kotlin

rbrott, 2020-12-16:
Tank PIDVA follower patched in 230cbf616ab166b25d8ce0e6d5f9cb6bca16278d.

closed 2020-12-16 by rbrott