acmerobotics/road-runner #69: robotToWheelVelocities for tankDrive

wangxdflight, 2021-02-22:
Is code here correct?
https://github.com/acmerobotics/road-runner/blob/2280420702e93765c5ed9873ce65d5f0e0411c4c/core/src/main/kotlin/com/acmerobotics/roadrunner/kinematics/TankKinematics.kt#L22

The input is the results after rotation from field velocity.
Does it need one more step (to figure out fieldPositionDelta) as the inverse of code at: https://github.com/acmerobotics/road-runner/blob/2280420702e93765c5ed9873ce65d5f0e0411c4c/core/src/main/kotlin/com/acmerobotics/roadrunner/kinematics/Kinematics.kt#L100 ?

            val deltaX = robotVel.x
            var rt = deltaX / sin(theta)  
            val r_minus_l = trackWidth * theta
            val r_plus_l = rt / (trackWidth / 2) * r_minus_l
            R = (r_plus_l + r_minus_l) / 2
            L = (r_plus_l - r_minus_l) / 2
            return listOf(L, R)

rbrott, 2021-02-22:
> Is this function correct?

I believe so. An error in such a crucial function would likely be obvious to tank users.
 
> The input is the results after rotation from field velocity.

The input should be "velocity of the robot in its reference frame." This is slightly abused in some places where velocity is substituted for "power" or local derivatives. This abuse is implicitly justified by linearity.

> Does it need one more step (to figure out fieldPositionDelta) as the inverse of code at [...]

The `robotToWheelVelocities()` function does not intend to be the inverse of that code.

wangxdflight, 2021-02-22:
Thanks for the quick response.
We do so Y direction correction issue in our tests, and trying to debug it.

Isn't relativeOdometryUpdate part of inverse kinematics, and TankPIDVAFollower doing forward kinematics based on target velocities and poses?

Before calculating "velocity of the robot in its reference frame" (where Y direction is always 0), shouldn't we get robot local velocities (where Y direction might not be 0, fieldPositionDelta in relativeOdometryUpdate())?

The results between with and without the step are similar most of the time, but sometimes do show large difference.

rbrott, 2021-02-22:
> Isn't relativeOdometryUpdate part of forward kinematics, and TankPIDVAFollower doing inverse kinematics based on target velocities and poses?

Yes.

> Before calculating "velocity of the robot in its reference frame" (where Y direction is always 0) in robotToWheelVelocities(), shouldn't we get robot local velocities (where Y direction might not be 0, fieldPositionDelta in relativeOdometryUpdate())?

Yes, that's correct. Inside `TankPIDVAFollower#internalUpdate()`, `targetVel` is the **field/global**-frame velocity computed from the path. This is handed to `Kinematics#fieldToRobotVelocity()` which produces a **robot/body**-frame velocity. This base robot-frame velocity is then altered according by the feedback control law. Later on, in `TankDrive', the robot-frame velocity computed by the follower is converted into wheel velocities.

This process is reversed in the localizer with some displacement-related subtleties.

wangxdflight, 2021-02-22:
"Later on, in `TankDrive', the robot-frame velocity computed by the follower is converted into wheel velocities."
I am having problem to understand this part.
I think we need one more step (as the inverse of code below) , before we can use this "robot-frame velocity" to obtain wheel velocities. This is in addition to the matrix rotation in Kinematics#fieldToRobotVelocity().

        val (sineTerm, cosTerm) = if (dtheta epsilonEquals 0.0) {
            1.0 - dtheta * dtheta / 6.0 to dtheta / 2.0
        } else {
            sin(dtheta) / dtheta to (1 - cos(dtheta)) / dtheta
        }
        val fieldPositionDelta = Vector2d(
            sineTerm * robotPoseDelta.x - cosTerm * robotPoseDelta.y,
            cosTerm * robotPoseDelta.x + sineTerm * robotPoseDelta.y
        )

rbrott, 2021-02-22:
Yeah, I can see why it seems like a step is missing. The key observation is that the "follower" process operates on velocities, while the "localizer" process operates on displacements. The `relativeOdometryUpdate()` function, in essence, implements the inverse, **integral** version of `fieldToRobotVelocity()`. You can see this in the derivation provided in section 10.2.4 of [this text](https://file.tavsys.net/control/controls-engineering-in-frc.pdf).

wangxdflight, 2021-02-23:
Thanks. Looks like the math model used in that text is more advanced and elegant.
My understanding has been mainly from https://chsftcrobotics.weebly.com/uploads/1/2/3/6/123696510/odometry.pdf
For localization, the final math equations are the same even they use different approaches.

closed 2021-02-23 by wangxdflight