acmerobotics/road-runner #75: FTC Dashboard Wifi Problem

SomeoneSketchySync, 2022-02-07:
We are facing this strange error in accessing the FTC Dashboard after the 7.1 upgrade. We wanted to retune RoadRunner because of a slight mechanical redesign and ended up with this error. We also updated to the new release of RoadRunner.

[image]

Thanks,
AP, Team 19643

rbrott, 2022-02-07:
Dash may be somehow disabled. Try re-enabling it first. If it still doesn't work, open [dev tools](https://developer.mozilla.org/en-US/docs/Learn/Common_questions/What_are_browser_developer_tools) and capture the network tab (specifically the websocket at the end) and any console output.

SomeoneSketchySync, 2022-02-07:
Could you please let me know how to re-enable the dashboard? Thanks for the speedy reply!

SomeoneSketchySync, 2022-02-07:
Thanks! I just figured it out and now it works. I reenabled it via the Driver Station where under the TeleOp section there was an option to Enable/Disable the dashboard.

closed 2022-02-07 by SomeoneSketchySync