acmerobotics/road-runner #77: [Q] - How do you use the return from Trajectory.velocity()

EddieDL, 2022-02-28:
Can anyone explain what exactly `Trajectory.velocity()` returns and how to use it properly?

I am trying to get the robots velocity at a specific point in the trajectory. Right now, I am simply doing `traj.velocity(t).getX()`. That seems to work, somewhat. My overall goal is to use road-runner's motion profiling to get a trajectory and generate a list of motion profile points (velocity, position, heading) that a hardware controller will then follow.

Any thoughts or suggestions would be appreciated.

amarcolini, 2022-02-28:
If you just need a motion profile, instead of generating an entire trajectory you can use `MotionProfileGenerator`'s methods, which return a `MotionProfile`. Calling `MotionProfile.get()` gives you a `MotionState` with position, velocity, acceleration, and jerk.

`Trajectory.velocity()` returns the pose velocity at the provided time. `Trajectory.velocity().getX()` returns just the x velocity. If you want the total translational velocity, use `Trajectory.velocity().norm()`, and if you want the angular velocity, use `Trajectory.velocity().getHeading()`.

EddieDL, 2022-02-28:
Thanks,

I will look into the `MotionProfilelGenerator's" properties. Maybe that is what I should be doing, but I liked the idea of being able to use the `TrajectoryBuilder` to easily create my profile.

EddieDL, 2022-02-28:
@amarcolini I looked at the `MotionProfileGenerator` and I think it is too raw.

Would it actually be `Trajectory.velocity(t).vec().norm()` for the total velocity?

amarcolini, 2022-02-28:
For the total translational velocity, yes.

EddieDL, 2022-02-28:
Another question. Is this relative to the field or the robot? In my case I am using tank drive and I am pretty sure `Trajectory.velocity(t).vec().norm()` will always be positive. I need the velocity of the robot from the robots perspective positive or negative. I can calculate it myself, but I assume there is a way to do this already.

rbrott, 2022-02-28:
All `Trajectory`/`Path` outputs are in field coordinates. You can transform to the robot frame using [these methods](https://github.com/acmerobotics/road-runner/blob/448a3adb2d2f821aed9e42998ef216ca4732f690/core/src/main/kotlin/com/acmerobotics/roadrunner/followers/HolonomicPIDVAFollower.kt#L52-L57).

EddieDL, 2022-02-28:
I wondered about those methods. Since I am using tank drive I am looking at the ones that are in `TankPIDVAFollower`, which I think might work for me. However, I am confused as to why it is using both the x and the y velocity. I was expecting it to only use one of these (y). Is the x velocity the turn velocity?

rbrott, 2022-02-28:
The transform code is actually the same for `TankPIDVAFollower`. The strafe velocity is actually the `y`-component of the robot-frame pose velocity. If heading tracks the path tangent, this velocity will always be zero. The turn velocity is the heading component.

rbrott, 2022-03-01:
Yes, that will give you the distance between the two **positions**. That is **not** the same as the distance between the two **poses**.

EddieDL, 2022-03-01:
If I want to get the distance between two poses can I do `pose1.minus(pose2)`? if so what does that actually give me? Do I just need to convert that to a robot relative pose and take the value of x?

rbrott, 2022-03-01:
I think you probably want the [pose error](https://github.com/acmerobotics/road-runner/blob/448a3adb2d2f821aed9e42998ef216ca4732f690/core/src/main/kotlin/com/acmerobotics/roadrunner/followers/HolonomicPIDVAFollower.kt#L59). That's essentially the difference you proposed with some normalization and a change of coordinates. (All of this is very ad hoc. Here's a better approach for extra credit: find the transform between the poses and "flatten" the coordinates as done [here](https://github.com/rbrott/road-runner2/blob/af053bed4063ee855df28642d1215d09572e685c/src/main/kotlin/com/acmerobotics/roadrunner2/Geometry.kt#L161).)

If you still want distance, you can apply a norm to the error. You need to choose a length scale and can then do a weighted sum of the vector norm and the angle magnitude.

EddieDL, 2022-03-01:
I apologize for not being clearer about what I am actually doing. I am trying to adapt Roadrunner for use in FRC, which can deal with path following very different than FTC. My reasoning for doing this is that we work in both programs and feel it would be beneficial to use the same path generation tool for both programs. I am not sure how much you know about FRC following, but I am going to provide a bit of background.

We are using Falcon 500's on our drivetrain which support motion profiling on the motor controller. Effectively you simply load up `TrajectoryPoint`'s into the motor controller and it has it's own internal PID to follow the position and heading (auxiliary position). Below is an example of the `TrajecotryPoint` that we have to build to make this work.
```
position = (rightEncoder + leftEncoder) / 2.0,
velocity = inchestoEnc(curVelPose.getX()) /10.0, // Velocity is in counts/100ms
auxiliaryPos = convertToUnits(curPose.GetHeading()) // The gyro we are using has special units
```
To make this work, we are currently doing the following logic to calculate the right and left encoder values that we expect.
```
Vector2d curLeft = generateVector(curPose, false);
Vector2d curRight = generateVector(curPose, true);

double curLeftSegment = prevLeft.distTo(curLeft);
double curRightSegment = prevRight.distTo(curRight);

rightEncoder += (Math.signum(point.velocity) * inchesToEnc(curRightSegment));
leftEncoder += (Math.signum(point.velocity) * inchesToEnc(curLeftSegment));
.
.
.
  private Vector2d generateVector(Pose2d pose, boolean right) {
    double direction = 1;
    if(right) {
      direction = -1;
    }

    double cos_angle = Math.cos(pose.getHeading() + (direction * (Math.PI / 2.0)));
    double sin_angle = Math.sin(pose.getHeading() + (direction * (Math.PI / 2.0)));

    double X = pose.getX() - halfTrackWidth * cos_angle;
    double Y = pose.getY() - halfTrackWidth * sin_angle;

    return new Vector2d(X, Y);
  }
```
I noticed while writing this that I think the `curPose` is field-oriented when we really need the robot-oriented pose. From what I can tell the `Kinematics` doesn't have a basic function to convert from field-oriented to robot-oriented. I am not sure how to do that, but that is something I can investigate

If I want to get "extra credit", how do I use the `Rotation2` class? I am not sure what the `real` and `imag` parameters are. Also, it looks like the `minus` function you referenced is logging. I am not sure if that is Okay or not, but thought I would mention it.

I did some reading on what exactly the norm of the vector is and I think I understand the angle magnitude. It isn't clear to me how I would choose a length scale (unless you just mean inches) or perform the weighted sum.

I really appreciate @rbrott and @amarcolini spending time to respond to this. I know this use case isn't what roadrunner was built for and it is possible our approach is completely wrong.

NoahBres, 2022-03-01:
@EddieDL Why don't you just try and stick with WPILib? It seems like a much less difficult endeavor to try and get WPILib working for FTC rather than contorting FTC tooling for FRC. WPILib has a much more active and mature developer community. In addition, there is work on this already with FTCLib, although I cannot speak on its maturity.

rbrott, 2022-03-01:
> We are using Falcon 500's on our drivetrain which support motion profiling on the motor controller.

Okay I think I know what's going on now. You want to sample position and velocity for individual wheels along with heading and angular velocity. In that case, the first suggestion of @amarcolini covers almost everything you need except position. Accumulating pairwise differences between sampled **wheel** positions is crude yet probably sufficient for this kind of following.

But this system is severely flawed as a means for tracking robot trajectories. You want a controller that assesses the pose/configuration error to achieve proper position control. 

> Why don't you just try and stick with WPILib? It seems like a much less difficult endeavor to try and get WPILib working for FTC rather than contorting FTC tooling for FRC.

The new trajectory functionality in WPILib probably closer to what RR offers by default. This seems closer to what Jaci's original Pathfinder library provides (just integrated into the motor controller). But doubtless the FRC community will be better able to help you.

> If I want to get "extra credit", how do I use the `Rotation2` class? I am not sure what the `real` and `imag` parameters are. Also, it looks like the `minus` function you referenced is logging. I am not sure if that is Okay or not, but thought I would mention it.

> I did some reading on what exactly the norm of the vector is and I think I understand the angle magnitude. It isn't clear to me how I would choose a length scale (unless you just mean inches) or perform the weighted sum.

This is mostly irrelevant to what you're trying to accomplish.

> WPILib has a much more active and mature developer community.

😢

EddieDL, 2022-03-02:
@amarcolini Your suggestion worked great. Thanks for the help.

@NoahBres As I understand it, WPILib didn't have anything similar to RR until recently (perhaps 2 years ago). Even if I did use that implementation, it would have similar issues to RR. It is meant to follow profiles on the Roborio not send trajectory points to the motor controllers. I have considered prioritizing the FRC programming paradigm. I prefer the command based approach of WPILib. However, the last time I looked at FTCLib it was not far enough along and I was already familiar with RR. I am going to keep an eye on it and eventually we will probably switch. Just not yet.

@rbrott I am sure what I am doing has some limitations, but I actually think it works phenomenal for tank drive with no deadwheel tracking, which is the standard for FRC. 

All: I feel like you don't give yourselves enough credit. Using RR has been a great experience for me and I think the documentation/quickstart is some of the best stuff I have ever seen.

Thanks for everything you do!

closed 2022-03-02 by EddieDL