acmerobotics/road-runner #84: All encoders must be attached to the same hub

ftcteam8645, 2023-01-28:
We have a fair amount of encoders on our robot and are using 3 wheel odometry.  When we try to run the ForwardPushTest, we get an error telling us all encoders much be attached to the same hub.  This is an odd error, but obviously teams will have many encoders attached to all ports and hubs.  Is this really a requirement?  In the past we put the encoder from the odometry pods into the encoder port of the drive motor.  When we changed the ThreeDeadWheelLocalizer to:

        par0 = new RawEncoder(hardwareMap.get(DcMotorEx.class, "leftFront"));
        par1 = new RawEncoder(hardwareMap.get(DcMotorEx.class, "rightBack"));
        perp = new RawEncoder(hardwareMap.get(DcMotorEx.class, "leftBack"));

Is this a bug or are we doing something wrong?  We tried moving the three wheel encoders to one hub, but still had the same error.

ftcteam8645, 2023-01-28:
Resolved

closed 2023-01-28 by ftcteam8645

rbrott, 2023-01-29:
The restriction is intended, and I'd like to remove it. The only obstacle is getting accurate timing of measurements. I found it easier to assume all encoders are on the same hub and can be read with a single bulk read command.