acmerobotics/road-runner #87: Issue in Section 2.1 of "Mobile Robot Kinematics for FTC" Paper

ItsTheChickenMan, 2023-02-06:
Section 2.1 claims that the velocity of any point on a robot is given by the equation

$$ \vec{v} = \frac{d\vec{r}}{dt} = (v_x - y \ \omega) \hat{\imath} + (v_y + x \ \omega) \hat{\jmath} $$

with $x$ and $y$ defined by the equation

$$ \vec{r} = x  \hat{\imath} + y  \hat{\jmath} $$

This makes the former equation erroneous, as $x$ and $y$ should be defined as the x and y components of the ${\vec{r}\ }'$ vector.  The former equation is using the expanded form of $\vec{\omega} \times \vec{r}$, not $\vec{\omega} \times {\vec{r}\ }'$.

This only affects this one instance, of course, and the correct equation is used for the rest of the paper.

rbrott, 2023-02-07:
If the first equation is erroneous, what would you replace it with? I'm confused what your exact objection is.

jkenney2, 2023-03-27:
**r** is defined as the position of some point on the robot in the world coordinate system. It is stated to have **i** and **j** components of x and y, respectively. 

**r'** is defined as a vector from the center of rotation of the robot to the same point on the robot. Its **i** and **j** components are not stated, but one could call them x' and y'.

So,

**r** = **R** + **r'** , where **R** is the vector from the origin of the world coordinate system to the center of rotation of the robot.

The equation for the velocity of the point on the robot would therefore be:

d**r**/dt = (v<sub>x</sub> - wy')**i** + (v<sub>y</sub> + wx')**j**

commit e0d00cb9 (2023-03-28): Fix typo (fixes #87)
commit 7ba1527a (2023-03-28): Fix typo (fixes #87)
commit 5b1d9c02 (2023-08-06): Fix typo (fixes #87)
closed 2023-03-28 by rbrott