acmerobotics/road-runner #90: Rotation2d.log() needs a better name

j5155, 2023-09-13:
https://github.com/acmerobotics/road-runner/blob/c3f4b1ae5b06637788a034d2fa9c645851f323d4/core/src/main/kotlin/com/acmerobotics/roadrunner/Geometry.kt#L99
This could just be me not being super experienced with math and stuff like this, but naming this function log made it much more difficult to intuitively find and not finding it was the cause of my team's field centric driving issues over the summer. Maybe something like .toRadians(), .asDouble(), or .normalized() would be better here instead?

bweis, 2023-09-13:
See the function header:

```
 * Advanced: Rotations in two dimensions comprise a Lie group referred to as SO(2). The terminology [exp] and [log]
 * comes from the Lie theory, and [this paper](https://arxiv.org/abs/1812.01537) gives a targeted exposition of the key
 * fundamentals.
```

rbrott, 2023-09-14:
> `Rotation2d.log()` needs a better name
> Maybe something like .toRadians(), .asDouble(), or .normalized() would be better here instead?

I'm open to making `asDouble()` an alias. That suggests `fromDouble()` as an `exp()` alias which seems fine.

> This could just be me not being super experienced with math and stuff like this

Not your fault. I've been focusing more on getting the design right/correct at this stage instead of making the library easy/approachable.

> not finding it was the cause of my team's field centric driving issues over the summer

I'm curious about this. How are you using `log()`?

j5155, 2023-09-14:
> How are you using `log()`?
Well, I wasn't...I had poseEstimate.heading.real instead
https://github.com/jdhs-ftc/offseason-rr1.0/blob/bb60dc047bfe89d4a8e524dfd302371b8d7dbc5b/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/TeleopFieldCentric.java#L106
(code based on the LRR example)

rbrott, 2023-09-18:
I think you could do something like this instead:
```java
Vector2d input = new Vector2d(
  -gamepad1.left_stick_y * speed,
  -gamepad1.left_stick_x * speed
);

if (fieldCentric) {
  input = drive.pose.heading.inverse().times(
    new Vector2d(-input.x, input.y));
}
```

commit 31cae325 (2023-11-26): Add Rotation exp/log aliases (fixes #90)

I intend for the geometry classes to have sound foundations while
also being easy to use in the common setting of building trajectories.
Most users won't be invoking methods directly on the geometry classes,
so I took the liberty of keeping technical Lie theory names.

I still like that decision overall, though I'm happy to enhance usability
by providing aliases. In this case, I settled on `fromDouble()` and
`toDouble()` for the `Rotation2d` `exp()`/`log()` functions.
closed 2023-11-26 by rbrott