acmerobotics/road-runner #97: Action building causes freezing

FalseMSP, 2024-01-15:
We encountered an error where the building of a trajectory causes the driver station to freeze.
`positionNearStack = new Pose2d(-12, -48, Math.toRadians(270));
Pose2d stack3 = new Pose2d(-12,-62,Math.toRadians(270));
// stack 3 is later assigned to the firstStackGrab
Action driveIntoFirstStack = drive.actionBuilder(positionNearStack)
                .splineTo(firstStackGrab.position,firstStackGrab.heading)
                .waitSeconds(2)
                .splineTo(positionNearStack.position,positionNearStack.heading)
                .build();`
Here is the link to the file inside the commit with errors: https://github.com/FTC7393/FtcRobotController/blob/110f013d0eeec0f219ea966f44720a91ff109637/TeamCode/src/main/java/org/firstinspires/ftc/teamcode/AutoSteve.java
Interesting note: it works for different stack positions (ie. Pose2d stack1 = new Pose2d(-36,-62,Math.toRadians(270));)

Thanks

rbrott, 2024-01-15:
Interesting, I can reproduce the problem on the latest version. The angular sampling step hangs because the integrand isn't smooth enough or is too small. 

The easy fix is to replace the splines with lines. The numerics are better all around, including in this computation. 

Separately, it might be nice to disable/loosen the epsilon for angular sampling. I'll have to think more about accommodating that in the API.

rbrott, 2024-03-03:
Should be mitigated now with a change of params.

closed 2024-03-03 by rbrott