acmerobotics/road-runner #100: Question of RR 0.5 and .splineTo Sequences

mjreiland, 2024-02-11:
Sorry if this is a stupid question.  We haven't yet upgraded to RR 1.x and are still on 0.5 maybe this is better on 1.x or we have implemented it incorrectly.

We have a set of Trajectories and inside a single trajectory multiple .splineTo commands that should be very smooth without any discontinuity.  We have played them through .meepmeep, they look smooth (no strange hooks or kinks) but the robot still comes to a stop between each spline segment.  Looking at the documentation it seems to show that if the splines are smooth it should daisy chain one to the next as a super spline.  Any suggestions?  I will note that the section we are having difficulty with has 3 splines which are all driven in reverse back to back.  setReversed(true) at the beginning.

I can post the actual positions and .meepmeep chart tomorrow to show.

rbrott, 2024-02-11:
The code with exact values will be very helpful.

mjreiland, 2024-02-12:
@rbrott Thank you for your prompt offer for assistance.  The students worked on it over the weekend and what it apparently was is that in the conversion from .lineToLinears to Splines, they combined multiple .splineTo's in a sequence that were originally in different trajectories, but what they did wrong was put .setReversed(true) before each one of them.  So in a single trajectory it looked like this:

.setReversed(true)
.splineTo(15,4)
.setReversed(true)
.splineTo(20,10)
.setReversed(true)
.splineTo(30,15)

instead of the correct way:
.setReversed(true)
.splineTo(15,4)
.splineTo(20,10)
.splineTo(30,15)
.setReversed(false)

Once the extra .setReversed were removed, the path was continuous and worked perfectly.  Thank you again.

closed 2024-02-12 by rbrott