acmerobotics/road-runner #101: How to us April Tag for re-localization

pgaston, 2024-03-04:
(Using meepmeep here - hopefully that isn't the issue...   Any help is appreciated!   I tried reddit, but didn't get any takers.  
 Hopefully I'm not too far out in left field...)

Two thoughts on how to integrate April Tags into our auto - using roadrunner 1.0...    (btw, rr + meepmeep looks great!)

1. In a large auto TrajectorySecuence, use addDisplacementMarker when we want to re-localize.    This doesn't work.   I'm assuming (?) that when the TrajectorySecuence did the 'build()' it baked in poses?    Or is this a meepmeep interaction?   When the new bot pose is set, meepmeep flashes the robot in the new position, then immediately continues ignoring the changed position.
`                // Drop off second one
                .splineTo(new Vector2d(aprilTagDropoff.getX(),aprilTagDropoff.getY()),aprilTagDropoff.getHeading())
                // Read AprilTag for re-localization
                .addDisplacementMarker(() -> {
                    // This runs after the last splineTo completes - we want to be mostly stopped for april tags to work
                    currentPose = getAprilTagPose()  // current pose if we don't see the AT
                    // Testing - re-localize enough so can see on meepmeep
                    Pose2d testPose = new Pose2d(currentPose.getX()-20,currentPose.getY(),currentPose.getHeading());
                    myBot.setPose(testPose);                 // relocalized - but not working
                })
`
2. Use multiple TrajectorySequence (s) - with the AT read between them.   Similar to 0.56 .   This isn't working for me - in the example below only the second/last TrajectorySequence is run.
`        myBot.followTrajectorySequence(tsAlpha);
        myBot.setPose(getAprilTagPose());       // will return currentPose if AT not found
        TrajectorySequence tsToDropOff = myBot.getDrive().trajectorySequenceBuilder(myBot.getPose())
                // Drop off second one
                .splineTo(new Vector2d(aprilTagDropoff.getX(),aprilTagDropoff.getY()),aprilTagDropoff.getHeading())
                .build();
        myBot.followTrajectorySequence(tsToDropOff);
`

pgaston, 2024-03-04:
Or I could just be lost...   meepmeep works well at least!

rbrott, 2024-03-04:
>  using roadrunner 1.0...

I'm confused because you mention RR 1.0 but then talk about trajectory sequences below which are a 0.5 quickstart feature. The API in your examples is also 0.5.

> (Using meepmeep here - hopefully that isn't the issue... Any help is appreciated! I tried reddit, but didn't get any takers. Hopefully I'm not too far out in left field...)

MeepMeep isn't a great place to experiment with relocalization. It only knows about the desired trajectory with its perfect poses.

> When the new bot pose is set, meepmeep flashes the robot in the new position, then immediately continues ignoring the changed position.

This is pretty much what I would expect.

Check out https://github.com/acmerobotics/road-runner-quickstart/issues/367 for a better starting point.

pgaston, 2024-03-04:
> Check out acmerobotics/road-runner-quickstart#367

Interesting.    Thought I was following 1.0.      Whoops!

As a mentor I’m new to FTC this year and Android Studio/java.     (ROS2
experience…)

Thanks for feedback!
P

pgaston, 2024-03-05:
Much better!

fyi - perhaps a few more words on the opening of the MeepMeep readme would
be better.   As in "This" is the current fork...      One (and I think I
did) interpreted that as that link leads one to the fork that support RR
1.0.     Maybe it's me - but I didn't see anything calling out versions...

Great work!   Love it.     Kids here built their own version last year -
barely works - author graduated - new coder spends 90% of their time trying
to understand half-baked code w/o comments - etc.    Next year we're
learning this!